mathiask88/node-snap7 #2: ReadMultiVars callback not triggering

Lailend, 2015-02-21:
the function `S7Client.ReadMultiVars(multiVars, [callback])` doesn't seem to trigger the callback. 
The method itself seems to be working fine in a blocking manner (without the callback). But when I specify the callback it simply never triggers.

Here a example of the issue:

    var items = [{
      Area: 132, WordLen: 4, DBNumber: 2, Start: 24, Amount: 1
    }, {
      Area: 132, WordLen: 6, DBNumber: 2, Start: 46, Amount: 1
    }];

    s7client.ReadMultiVars(items, function (err, res) {
      console.log('callback triggered');
      if (err) {
        log.error(err);
      } else {
        log.info(res);
        processSessionReadItems(res);
      }
    });

Am I simply doing something wrong or isn't it implemented yet? The documentation about this function isn't documented either, so my excuses if it isn't ready yet.

mathiask88, 2015-02-21:
Hi, no you are not doing something wrong, the async part of this function is not implemented yet. I got some time this weekend and I think I can add this. And yes the docs are a bit messy and quite a spot of work :/

mathiask88, 2015-03-05:
@Lailend Sorry for the delay. I added all the missing functionality(not only ReadMultiVars) in the client and now I have to update the docs. I hope I can push the changes to master within the next week.

Lailend, 2015-03-07:
@mathiask88 No problems! I am already happy that you take the effort to do this. So thanks so much really appreciate it!

Is the `S7Client.ReadSZL()` also by any chance implemented in the next push?

mathiask88, 2015-03-07:
> Is the S7Client.ReadSZL() also by any chance implemented in the next push?

Yes, every client function should work. I pushed my WIP to master so you can test a little bit, before release and while I am updating the docs. To do so you can add the repo to your ```package.json```(https://docs.npmjs.com/files/package.json#git-urls-as-dependencies). For your information: ```ReadSZL(ID, Index, [callback])``` and returns a buffer.

Lailend, 2015-03-07:
Awesome really impressed. Really gratefull for the fast help and WIP push. 
Will defenitely try a few of these new functions out!

mathiask88, 2015-03-07:
I am happy about feedback ;)

celevra, 2015-04-20:
hi,

thank you very much for this library.
isn't it possible to use s7client.S7AreaDB and s7client.S7WLBytes in ReadMultiVars as like this:

```
var items = [{
  Area: s7client.S7AreaDB, WordLen: s7client.S7WLBytes, DBNumber: 1, Start: 0, Amount: 1
}, {
  Area: s7client.S7AreaDB, WordLen: s7client.S7WLBytes, DBNumber: 1, Start: 2, Amount: 1
}];
```

regards

mathiask88, 2015-04-20:
hi @celevra,
yes this is possible and recommended.

celevra, 2015-04-22:
thank you for your fast answer, but if i try it like this:

```
var items = [{
  Area: s7client.S7AreaDB, WordLen: s7client.S7WLBytes, DBNumber: 1, Start: 0, Amount: 1
}, {
  Area: s7client.S7AreaDB, WordLen: s7client.S7WLBytes, DBNumber: 1, Start: 2, Amount: 1
}];

s7client.ReadMultiVars(items, function (err, res) {
		if(err) return console.log(' >> ReadMultiVars failed. Code #' + err + ' - ' + s7client.ErrorText(err));

        console.log(res);
});
```

i get

```
/node/s7/snap.js:32
        s7client.ReadMultiVars(items, function (err, res) {
                 ^
TypeError: Wrong argument structure
```

what would be the correct argument structure?

i plan to use your library intensive so, if you don't mind, i would create an examples directory to show others how to use the function calls

mathiask88, 2015-04-22:
Yes, the error is correct because you got a typo in ```s7client.S7WLByte``` . I didn't see it either two days ago, sorry for that! I am currently updating the readme, but I don't got much time at the moment. An example directory sounds great.

celevra, 2015-04-22:
ah ok, now it's working. Thank you!
overall in the doku is s7client.S7WLBytes ;-)

closed 2015-06-29 by mathiask88