mathiask88/node-snap7 #3: Write, single bit example

cropii, 2015-03-18:
Hello i read bite like this, bud i can't change it and write back.

i got 100.1 
100.2 
they are false i need change it to true and write it back.
s7client.MBRead(100, 1, function(err, buffer) {
if(err) return console.log(' >> MBRead failed. Code #' + err + ' - ' + s7client.ErrorText(err));
			var bit=(buffer[0] & 0x01)!=0;

			console.log(bit); // say's false.
});

Some one please help me to write function that will change bit and write it back

mathiask88, 2015-03-18:
Hi,

if you want to write only a single bit you have to use
```S7Client.WriteArea(area, dbNumber, start, amount, wordLen, buffer, [callback])```
with ```wordLen=S7WLBit```.

For example if you want to change ```M100.1``` to true:
```javascript
var snap7 = require('node-snap7');

var s7client = new snap7.S7Client();
s7client.ConnectTo('127.0.0.1', 0, 1, function(err) {
    if(err)
        return console.log(' >> Connection failed. Code #' + err + ' - ' + s7client.ErrorText(err));

    // Change M100.1 to true
    s7client.WriteArea(s7client.S7AreaMK, 0, (100*8+1), 1, s7client.S7WLBit, new Buffer([0x01]), function(err) {
        if(err)
            return console.log(' >> WriteArea failed. Code #' + err + ' - ' + s7client.ErrorText(err));
    });
});
```

cropii, 2015-03-18:
thank you!!! it's work's !

bud now when i read i any way get false in my function maybe i do somting wrong when read bit?

mathiask88, 2015-03-18:
In your given example you read the full merker byte M100 but you test for the first bit with ```0x01``` but M100.1 is the second bit and M100.2 the third. So you have to do ```var bit = !!(buffer[0] & 0x02)``` for M100.1 and ```var bit = !!(buffer[0] & 0x04)``` for M100.2.

cropii, 2015-03-18:
And again thank you!

closed 2015-03-18 by cropii