mathiask88/node-snap7 #5: Build for win32

smiling-Jack, 2015-04-23:
Hallo
ist es möglich irgendwo ein win32 build zu downloaden ?

Ich habe nur Win64 und probleme eine 32bit Version selber zu bauen.

Können nicht die Test builds downloadbar gemacht werden?

mathiask88, 2015-04-24:
Hi,
das Problem bei fertigen builds ist, dass die builds für eine bestimmte node/iojs version compiliert sind und daher auch nur mit dieser nutzbar sind. Wo liegt denn das Problem für 32bit zu compilieren? Bei mir funktioniert ```npm rebuild --target_arch=ia32```.

smiling-Jack, 2015-04-24:
ah rebuild über npm hatte ich noch nicht probiert hatte es er Visualstudio versucht bin dort aber an der IDE gescheitert :-(

ich schreibe gerade eine Adaption für S7 an [iobroker](http://iobroker.net)
und da ist es nicht schön wenn die Windowsnutzer eine Visualstudio Installation haben müssen.

für die linux/arm Leute ist es ja kein Problem. Daher wollte ich für Windows ein build mitliefern und nur unter Linux eine npm Installation machen. 

Wenn ich jetzt ein build mit node.js 0.12 mache ist das dann garnicht abwärtskompatibel?

mathiask88, 2015-04-24:
>> und da ist es nicht schön wenn die Windowsnutzer eine Visualstudio Installation haben müssen.

Da führt wohl kein weg dran vorbei. Native node.js addons brauchen eine toolchain auf dem pc und werden gegen die installierte node.js Version compiliert. Daher gibt es ja auch den befehl ```npm rebuild```. Wenn man node.js updated muss man auch einen rebuild ausführen bei den Abhängigkeiten.

closed 2015-06-29 by mathiask88