mathiask88/node-snap7 #7: On Mac OS, callback function raises "Abort trap 6" error.

lsibilla, 2015-07-05:
Using this node app : https://www.dropbox.com/s/9a7benbp2x0nws5/node-snap7-test.zip?dl=0

Launch the command "npm install && node app.js" will lead to this output :
```
$ npm install
npm WARN package.json @ No description
npm WARN package.json @ No repository field.
npm WARN package.json @ No README data
\
> node-snap7@0.2.0 install /Users/lsibilla/Dropbox (Personnelle)/Ménage/Domotique/node-snap7-test/node_modules/node-snap7
> node-gyp rebuild

  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_msgsock.o
In file included from ../deps/snap7/src/sys/snap_msgsock.cpp:27:
../deps/snap7/src/sys/snap_msgsock.h:210:22: warning: private field 'RawPinger'
      is not used [-Wunused-private-field]
    PRawSocketPinger RawPinger;
                     ^
../deps/snap7/src/sys/snap_msgsock.h:211:10: warning: private field 'RawAvail'
      is not used [-Wunused-private-field]
    bool RawAvail;
         ^
2 warnings generated.
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_sysutils.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_tcpsrvr.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_threads.o
In file included from ../deps/snap7/src/sys/snap_threads.cpp:27:
In file included from ../deps/snap7/src/sys/snap_threads.h:41:
../deps/snap7/src/sys/unix_threads.h:184:10: warning: private field
      'FCreateSuspended' is not used [-Wunused-private-field]
    bool FCreateSuspended;
         ^
1 warning generated.
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_client.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_isotcp.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_partner.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_peer.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_server.o
../deps/snap7/src/core/s7_server.cpp:345:6: warning: variable 'DBNum' is used
      uninitialized whenever 'if' condition is false [-Wsometimes-uninitialized]
        if (ReqItemPar->Area==S7AreaDB)
            ^~~~~~~~~~~~~~~~~~~~~~~~~~
../deps/snap7/src/core/s7_server.cpp:350:30: note: uninitialized use occurs here
        P=GetArea(ReqItemPar->Area, DBNum);
                                    ^~~~~
../deps/snap7/src/core/s7_server.cpp:345:2: note: remove the 'if' if its
      condition is always true
        if (ReqItemPar->Area==S7AreaDB)
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../deps/snap7/src/core/s7_server.cpp:328:12: note: initialize the variable
      'DBNum' to silence this warning
        word DBNum;
                  ^
                   = 0
../deps/snap7/src/core/s7_server.cpp:578:6: warning: variable 'DBNum' is used
      uninitialized whenever 'if' condition is false [-Wsometimes-uninitialized]
        if (ReqItemPar->Area==S7AreaDB)
            ^~~~~~~~~~~~~~~~~~~~~~~~~~
../deps/snap7/src/core/s7_server.cpp:583:30: note: uninitialized use occurs here
        P=GetArea(ReqItemPar->Area, DBNum);
                                    ^~~~~
../deps/snap7/src/core/s7_server.cpp:578:2: note: remove the 'if' if its
      condition is always true
        if (ReqItemPar->Area==S7AreaDB)
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../deps/snap7/src/core/s7_server.cpp:564:15: note: initialize the variable
      'DBNum' to silence this warning
    word DBNum;
              ^
               = 0
In file included from ../deps/snap7/src/core/s7_server.cpp:26:
../deps/snap7/src/core/s7_server.h:122:15: warning: private field 'Buffer' is
      not used [-Wunused-private-field]
    TS7Buffer Buffer;
              ^
3 warnings generated.
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_text.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_micro_client.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/lib/snap7_libmain.o
  LIBTOOL-STATIC Release/snap7.a
  CXX(target) Release/obj.target/node_snap7/src/node_snap7.o
  CXX(target) Release/obj.target/node_snap7/src/node_snap7_client.o
cat pac  CXX(target) Release/obj.target/node_snap7/src/snap7.o
ka  SOLINK_MODULE(target) Release/node_snap7.node
  SOLINK_MODULE(target) Release/node_snap7.node: Finished
node-snap7@0.2.0 node_modules/node-snap7
├── bindings@1.2.1
└── nan@1.8.4
$ node app.js
	Abort trap: 6
```

mathiask88, 2015-07-05:
What's your node version?

lsibilla, 2015-07-05:
It is version 0.10.15.

mathiask88, 2015-07-05:
Can you run
```
gdb --args node app.js
#and in (gdb)
run
backtrace full
```
and post the backtrace?

mathiask88, 2015-07-05:
@lsibilla And can you try to use the current 0.10.39er version? 0.10.15 is nearly 2 years old and there was a bug in node that led to such an error on OSX.

lsibilla, 2015-07-05:
Thank you for your support.

I updated to 0.10.39.

The result is still the same. Here is the debugger output:
```
$ node --version
v0.10.39
$ lldb node app.js
(lldb) target create "node"
Current executable set to 'node' (x86_64).
(lldb) settings set -- target.run-args  "app.js"
(lldb) run
Process 8127 launched: '/usr/local/bin/node' (x86_64)
Process 8127 stopped
* thread #1: tid = 0x49ddf13, 0x00007fff96589286 libsystem_kernel.dylib`__pthread_kill + 10, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff96589286 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff96589286 <+10>: jae    0x7fff96589290            ; <+20>
    0x7fff96589288 <+12>: movq   %rax, %rdi
    0x7fff9658928b <+15>: jmp    0x7fff96584c53            ; cerror_nocancel
    0x7fff96589290 <+20>: retq   
(lldb) bt all
* thread #1: tid = 0x49ddf13, 0x00007fff96589286 libsystem_kernel.dylib`__pthread_kill + 10, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff96589286 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff8fe2b42f libsystem_pthread.dylib`pthread_kill + 90
    frame #2: 0x00007fff8df6db53 libsystem_c.dylib`abort + 129
    frame #3: 0x0000000100448877 node`uv_mutex_unlock + 20
    frame #4: 0x00000001029cfdcf node_snap7.node`node_snap7::IOWorker::HandleOKCallback(this=0x0000000100d03ca0) + 1635 at node_snap7_client.cpp:912
    frame #5: 0x00000001029d0c3f node_snap7.node`NanAsyncWorker::WorkComplete(this=0x0000000100d03ca0) + 47 at nan.h:1616
    frame #6: 0x00000001029d112a node_snap7.node`NanAsyncExecuteComplete(req=<unavailable>) + 18 at nan.h:1791
    frame #7: 0x0000000100448f6d node`uv__work_done + 168
    frame #8: 0x000000010043fc88 node`uv__async_event + 64
    frame #9: 0x000000010043fe0c node`uv__async_io + 136
    frame #10: 0x000000010044c8e4 node`uv__io_poll + 1503
    frame #11: 0x0000000100440251 node`uv_run + 239
    frame #12: 0x00000001003f9edc node`node::Start(int, char**) + 365
    frame #13: 0x0000000100001434 node`start + 52

  thread #2: tid = 0x49ddf33, 0x00007fff965844de libsystem_kernel.dylib`mach_msg_trap + 10
    frame #0: 0x00007fff965844de libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff9658364f libsystem_kernel.dylib`mach_msg + 55
    frame #2: 0x00007fff93071eb4 CoreFoundation`__CFRunLoopServiceMachPort + 212
    frame #3: 0x00007fff9307137b CoreFoundation`__CFRunLoopRun + 1371
    frame #4: 0x00007fff93070bd8 CoreFoundation`CFRunLoopRunSpecific + 296
    frame #5: 0x00007fff93128671 CoreFoundation`CFRunLoopRun + 97
    frame #6: 0x000000010044b3de node`uv__cf_loop_runner + 67
    frame #7: 0x000000010043f85a node`uv__thread_start + 25
    frame #8: 0x00007fff8fe29268 libsystem_pthread.dylib`_pthread_body + 131
    frame #9: 0x00007fff8fe291e5 libsystem_pthread.dylib`_pthread_start + 176
    frame #10: 0x00007fff8fe2741d libsystem_pthread.dylib`thread_start + 13

  thread #3: tid = 0x49ddf34, 0x00007fff9658a232 libsystem_kernel.dylib`kevent64 + 10, queue = 'com.apple.libdispatch-manager'
    frame #0: 0x00007fff9658a232 libsystem_kernel.dylib`kevent64 + 10
    frame #1: 0x00007fff9291cd91 libdispatch.dylib`_dispatch_mgr_invoke + 247
    frame #2: 0x00007fff9291ca6a libdispatch.dylib`_dispatch_mgr_thread + 52

  thread #4: tid = 0x49ddf35, 0x00007fff9658451a libsystem_kernel.dylib`semaphore_wait_trap + 10, name = 'SamplerThread'
    frame #0: 0x00007fff9658451a libsystem_kernel.dylib`semaphore_wait_trap + 10
    frame #1: 0x00000001003eff28 node`v8::internal::MacOSSemaphore::Wait() + 24
    frame #2: 0x00000001002fe189 node`v8::internal::RuntimeProfilerRateLimiter::SuspendIfNecessary() + 121
    frame #3: 0x00000001003efff8 node`v8::internal::SamplerThread::Run() + 88
    frame #4: 0x00000001003ef7cc node`v8::internal::ThreadEntry(void*) + 60
    frame #5: 0x00007fff8fe29268 libsystem_pthread.dylib`_pthread_body + 131
    frame #6: 0x00007fff8fe291e5 libsystem_pthread.dylib`_pthread_start + 176
    frame #7: 0x00007fff8fe2741d libsystem_pthread.dylib`thread_start + 13

  thread #5: tid = 0x49ddf36, 0x00007fff96589136 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #0: 0x00007fff96589136 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x00007fff8fe29e0c libsystem_pthread.dylib`_pthread_cond_wait + 693
[... 13 lines cut ...]
    frame #5: 0x00007fff8fe29268 libsystem_pthread.dylib`_pthread_body + 131
    frame #6: 0x00007fff8fe291e5 libsystem_pthread.dylib`_pthread_start + 176
    frame #7: 0x00007fff8fe2741d libsystem_pthread.dylib`thread_start + 13

  thread #7: tid = 0x49ddf38, 0x00007fff96589136 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #0: 0x00007fff96589136 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x00007fff8fe29e0c libsystem_pthread.dylib`_pthread_cond_wait + 693
    frame #2: 0x0000000100448a37 node`uv_cond_wait + 9
    frame #3: 0x00000001004492bb node`worker + 207
    frame #4: 0x000000010043f85a node`uv__thread_start + 25
    frame #5: 0x00007fff8fe29268 libsystem_pthread.dylib`_pthread_body + 131
    frame #6: 0x00007fff8fe291e5 libsystem_pthread.dylib`_pthread_start + 176
    frame #7: 0x00007fff8fe2741d libsystem_pthread.dylib`thread_start + 13

  thread #8: tid = 0x49ddf39, 0x00007fff96589136 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #0: 0x00007fff96589136 libsystem_kernel.dylib`__psynch_cvwait + 10
    frame #1: 0x00007fff8fe29e0c libsystem_pthread.dylib`_pthread_cond_wait + 693
    frame #2: 0x0000000100448a37 node`uv_cond_wait + 9
    frame #3: 0x00000001004492bb node`worker + 207
    frame #4: 0x000000010043f85a node`uv__thread_start + 25
    frame #5: 0x00007fff8fe29268 libsystem_pthread.dylib`_pthread_body + 131
    frame #6: 0x00007fff8fe291e5 libsystem_pthread.dylib`_pthread_start + 176
    frame #7: 0x00007fff8fe2741d libsystem_pthread.dylib`thread_start + 13
(lldb) 
```

mathiask88, 2015-07-05:
Should be fixed with https://github.com/mathiask88/node-snap7/commit/f3be865dc27dd8cfcdefa0e453ee1f6b4391d19d. Thanks for your help!

lsibilla, 2015-07-05:
Great ! It works well. Thank you very much.

closed 2015-07-05 by lsibilla