mathiask88/node-snap7 #8: Cannot read on Mac OS

lsibilla, 2015-07-05:
Hello,

I have another issue on Mac OS. When running this code
```
var snap7 = require('node-snap7');

s7client = global.s7client = new snap7.S7Client();

s7client.SetConnectionParams("192.168.1.30", 01 << 8, 02 << 8);
s7client.Connect(function(err) {
    if(err) {
        return console.log(' >> Connection failed. Code #' + err + ' - ' + s7client.ErrorText(err));
    } else {
        s7client.ReadArea(s7client.S7AreaPA, 1, 0, 8*8, 1, function(err, dataAreaPA) {
            if (err) 
                console.log(' >> Read failed. Code#' + err + ' - ' + s7client.ErrorText(err));
        });
    }
});
```

I get this error:

```
$ node app.js
 >> Read failed. Code#10485760 - CPU : Invalid Transport size
```

This issue was not present when using the branch lsibilla/node-snap7.

mathiask88, 2015-07-05:
I cant imagine that this worked before. You can only read one bit inside a word if you use S7WLBit, so the amount must be 1.

mathiask88, 2015-07-05:
JFYI the S7 protocol does not allow to read more than one bit at a time. If you want to do this you have to read the whole bytes and extract your bits.

lsibilla, 2015-07-06:
Ok. I guess bda56fb045a25b39d0bc6eb37f39b78ca89600ac on deps/snap7/src/core/s7_micro_client.cpp screwed it up.

I adjusted my code to read bytes instead of bits and it works just fine. Thank you !

closed 2015-07-06 by lsibilla

mathiask88, 2015-07-06:
No problem, thank you for making this project better :) BTW I published a new version(0.2.0) on npm so that you can use proper versions in package.json instead of the master branch.