mathiask88/node-snap7 #9: Installation

rnaus, 2015-07-17:
Hello,
 I'm beginner in Javascript. I found library Node-snap7 to connect to PLC Siemens S7 and I want to use it in my simple programm in Javascript. I made instalation throught 
 npm install node-snap7 with this result : 

C:\Users\rnaus\node_modules\node-snap7>if not defined npm_config_node_gyp (node
 "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_mod
 ules\node-gyp\bin\node-gyp.js" rebuild ) else (node rebuild )
 Building the projects in this solution one at a time. To enable parallel build,
 please add the "/m" switch.
 snap_msgsock.cpp
 snap_sysutils.cpp
 snap_tcpsrvr.cpp
 snap_threads.cpp
 ..\deps\snap7\src\sys\snap_msgsock.cpp(109): warning C4996: 'inet_addr': Use in
 et_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to di
 sable deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\sna
 p7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1850) :
 see declaration of 'inet_addr'
 ..\deps\snap7\src\sys\snap_msgsock.cpp(125): warning C4996: 'inet_ntoa': Use in
 et_ntop() or InetNtop() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to di
 sable deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\sna
 p7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1868) :
 see declaration of 'inet_ntoa'
 ..\deps\snap7\src\sys\snap_msgsock.cpp(786): warning C4996: 'inet_addr': Use in
 et_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to di
 sable deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\sna
 p7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1850) :
 see declaration of 'inet_addr'
 ..\deps\snap7\src\sys\snap_msgsock.cpp(852): warning C4996: 'inet_addr': Use in
 et_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to di
 sable deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\sna
 p7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1850) :
 see declaration of 'inet_addr'
 s7_client.cpp
 s7_isotcp.cpp
 s7_partner.cpp
 s7_peer.cpp
 ..\deps\snap7\src\core\s7_partner.cpp(118): warning C4996: 'inet_ntoa': Use ine
 t_ntop() or InetNtop() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to dis
 able deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\snap
 7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1868) :
 see declaration of 'inet_ntoa'
 ..\deps\snap7\src\core\s7_partner.cpp(445): warning C4996: 'inet_addr': Use ine
 t_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to dis
 able deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\snap
 7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1850) :
 see declaration of 'inet_addr'
 ..\deps\snap7\src\core\s7_partner.cpp(446): warning C4996: 'inet_addr': Use ine
 t_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to dis
 able deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\snap
 7.vcxproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1850) :
 see declaration of 'inet_addr'
 s7_server.cpp
 s7_text.cpp
 s7_micro_client.cpp
 snap7_libmain.cpp
 ..\deps\snap7\src\core\s7_text.cpp(94): warning C4996: 'inet_ntoa': Use inet_nt
 op() or InetNtop() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable
 deprecated API warnings [C:\Users\rnaus\node_modules\node-snap7\build\snap7.vc
 xproj]
 C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock2.h(1868) :
 see declaration of 'inet_ntoa'
 snap7.vcxproj -> C:\Users\rnaus\node_modules\node-snap7\build\Release\\snap7.
 lib
 node_snap7.cpp
 node_snap7_client.cpp
 snap7.cpp
 C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xlocale(337): wa
 rning C4530: C++ exception handler used, but unwind semantics are not enabled.
 Specify /EHsc (..\src\node_snap7_client.cpp) [C:\Users\rnaus\node_modules\node-
 snap7\build\node_snap7.vcxproj]
 C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xlocale(337): wa
 rning C4530: C++ exception handler used, but unwind semantics are not enabled.
 Specify /EHsc (..\src\node_snap7.cpp) [C:\Users\rnaus\node_modules\node-snap7\b
 uild\node_snap7.vcxproj]
 C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xlocale(337): wa
 rning C4530: C++ exception handler used, but unwind semantics are not enabled.
 Specify /EHsc (..\src\snap7.cpp) [C:\Users\rnaus\node_modules\node-snap7\build\
 node_snap7.vcxproj]
 Creating library C:\Users\rnaus\node_modules\node-snap7\build\Release\node
 _snap7.lib and object C:\Users\rnaus\node_modules\node-snap7\build\Release\no
 de_snap7.exp
 Generating code
 Finished generating code
 node_snap7.vcxproj -> C:\Users\rnaus\node_modules\node-snap7\build\Release\\n
 ode_snap7.node
node-snap7@0.2.1 node_modules\node-snap7
 ├── bindings@1.2.1
 └── nan@1.8.4 

I'm not sure that installation is correct.
I tried to use example (I made file.js and I tried Open) 

// JavaScript Document
 var snap7 = require('node-snap7'); 

var s7client = new snap7.S7Client();
 s7client.ConnectTo('192.168.1.12', 0, 1, function(err) {
 if(err)
 return console.log(' >> Connection failed. Code #' + err + ' - ' + s7client.ErrorText(err)); 

// Read the first byte from PLC process outputs...
 s7client.ABRead(0, 1, function(err, res) {
 if(err)
 return console.log(' >> ABRead failed. Code #' + err + ' - ' + s7client.ErrorText(err)); 

// ... and write it to stdout
 console.log(res)
 });
 }); 

but in line var snap7 = require ('node-snap7');..... I have error message Microsoft JScript runtime error code 800A138F, object expected. 
 Can you help me ? 

Radek

mathiask88, 2015-07-17:
You know that this is a package for https://nodejs.org/ runtime and not a browser library?

rnaus, 2015-07-17:
I install nodejs. But I don't know where  I can try your example.

mathiask88, 2015-07-17:
Ok, put the example in the folder where you ran `npm install node-snap7` and then execute the example with `node example.js` from the command line. Remeber to change the IP address to the one from your PLC.

rnaus, 2015-07-17:
ok, script is working. But connection is fault. I tried ping - it si ok. Message is Connection failed Code #665420 - ISO: An error occured during recv TCP: Connection timeout.

rnaus, 2015-07-17:
I think that I made wrong number slot. Now is answer <Buffer 00>.

rnaus, 2015-07-17:
answer is "Buffer 00"

mathiask88, 2015-07-17:
Look at the https://github.com/mathiask88/node-snap7#connect-to and make sure ´rack´ and ´slot´ is correct.

Edit: Ah I see, you found it :)

rnaus, 2015-07-17:
ok, thank you for our answers.

rnaus, 2015-07-17:
your - :-)

closed 2015-07-18 by mathiask88

Kortenbach, 2018-05-22:
Can you please tell everybody what you have changed in order to get it to work properly?