mathiask88/node-snap7 #11: Build fails on non-root linux user

teropes, 2015-10-19:
I'm getting the following build error when installing node-snap7. Im operating on Centos 6 and trying to install as non-root user. Running on node 0.12.7, also tried with 0.10.40. Installing as root works. Is there any way to avoid installing this package as root?

> node-gyp rebuild
Traceback (most recent call last):
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 18, in <module>
    sys.exit(gyp.script_main())
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 533, in script_main
    return main(sys.argv[1:])
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 526, in main
    return gyp_main(args)
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 502, in gyp_main
    params, options.check, options.circular_check)
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 129, in Load
    params['parallel'], params['root_targets'])
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 2736, in Load
    variables, includes, depth, check, True)
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 386, in LoadTargetBuildFile
    includes, True, check)
  File "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 238, in LoadOneBuildFile
    None)
  File "binding.gyp", line 1
    {
     ^
SyntaxError: invalid syntax
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:355:16)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 2.6.32-504.30.3.el6.x86_64
gyp ERR! command "node" "/home/sysuser/.nvm/versions/node/v0.12.7/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /data/my-project/node_modules/node-snap7
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
npm ERR! Linux 2.6.32-504.30.3.el6.x86_64
npm ERR! argv "/home/sysuser/.nvm/versions/node/v0.12.7/bin/node" "/home/sysuser/.nvm/versions/node/v0.12.7/bin/npm" "install" "node-snap7"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE
npm ERR! node-snap7@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-snap7@0.2.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the node-snap7 package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-snap7
npm ERR! There is likely additional logging output above.
npm ERR! Please include the following file with any support request:
npm ERR!     /data/my-project/npm-debug.log

mathiask88, 2015-10-20:
Hi, sry for the late response. Can you check your python version that is installed for the non-root user? 2.7.3 or newer(not 3.x) is necessary for node-gyp.

teropes, 2015-10-20:
python --version outputs 2.7.8. My /usr/local/bin/python is symlink to /usr/local/bin/python2.7 and I havent got issues with building other packages in this project such as oracledb. I also tried with my other development machine which runs Oracle Linux 6.5 and got same issue there.

mathiask88, 2015-10-20:
Can you check if the binding.gyp contains CRLF or LF line endings? I'm on my phone right now. I'm developing under Windows and maybe the CRLF slipped in.

teropes, 2015-10-20:
My text editor on my dev machine (Sublime Text 3) shows Line Endings as "Windows" so that may indeed be the issue.

mathiask88, 2015-10-20:
Yes, there is something wrong indeed. On Github line endings are ok because of git's autoCRLF I think but npm uploaded the DOS/Windows line endings :( I'm about to release a new version anyway and will have a special look on LE this time. You can try install from github (e.g. `npm install mathiask88/node-snap7`) or wait until the new version is published. Thanks for your report!

mathiask88, 2015-10-20:
@teropes I published a new release to npm that fixed the line endings and is compatible to node 4.x. Would be cool if you can test it and report back.

teropes, 2015-10-21:
node-snap7 0.2.2 now builds properly also as non-root user in both of my Centos / Oracle Linux dev machines and no other issues detected so far. This is now fixed, cheers!

closed 2015-10-21 by teropes