mathiask88/node-snap7 #13: Install on win32 - build error

yasdfgr, 2016-01-14:
I tried to install with nmp install node-snap7 and get following errors. Whats wrong?
Python 2.7 and VisualStudio Express 2010 C++ is installed.
    
> node-snap7@0.2.2 install C:\Documents and Settings\Administrator\node_modules\node-snap7
> node-gyp rebuild

C:\Documents and Settings\Administrator\node_modules\node-snap7>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node  rebuild ) 
  snap_msgsock.cpp
  snap_sysutils.cpp
  snap_tcpsrvr.cpp
  snap_threads.cpp
  s7_client.cpp
  s7_isotcp.cpp
  s7_partner.cpp
  s7_peer.cpp
  s7_server.cpp
  s7_text.cpp
  s7_micro_client.cpp
  snap7_libmain.cpp
  win_delay_load_hook.c
  snap7.vcxproj -> C:\Documents and Settings\Administrator\node_modules\node-snap7\build\Release\\snap7.lib
  node_snap7.cpp
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(336): error C2988: unrecognizable template declaration/definition [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(336): error C2059: syntax error : 'using' [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(469): error C2988: unrecognizable template declaration/definition [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(469): error C2059: syntax error : 'using' [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(472): error C2332: 'enum' : missing tag name [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(472): error C3306: 'v8::<unnamed-tag>': unnamed class template is not allowed [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(472): error C2236: unexpected 'class' 'v8::WeakCallbackType'. Did you forget a ';'? [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(472): error C3381: 'v8::WeakCallbackType' : assembly access specifiers are only available in code compiled with a /clr option [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(580): error C2061: syntax error : identifier 'WeakCallbackType' [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(646) : see reference to class template instantiation 'v8::PersistentBase<T>' being compiled
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(641): error C2253: 'PersistentBase<T>' : pure specifier or abstract override specifier only allowed on virtual function [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(848): error C2253: 'Global<T>' : pure specifier or abstract override specifier only allowed on virtual function [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(851) : see reference to class template instantiation 'v8::Global<T>' being compiled
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(856): error C2988: unrecognizable template declaration/definition [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(856): error C2059: syntax error : 'using' [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(919): error C2989: 'v8::HandleScope' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(319) : see declaration of 'v8::HandleScope'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(953): error C2989: 'v8::EscapableHandleScope' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(135) : see declaration of 'v8::EscapableHandleScope'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(983): error C2989: 'v8::Data' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(74) : see declaration of 'v8::Data'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(1117): error C2989: 'v8::Script' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(96) : see declaration of 'v8::Script'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(1556): error C2989: 'v8::StackTrace' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(101) : see declaration of 'v8::StackTrace'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(1619): error C2989: 'v8::StackFrame' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(100) : see declaration of 'v8::StackFrame'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(2028): error C2989: 'v8::Value' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(108) : see declaration of 'v8::Value'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(2034): error C2989: 'v8::Primitive' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(93) : see declaration of 'v8::Primitive'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(2049): error C2989: 'v8::Boolean' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(70) : see declaration of 'v8::Boolean'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(2069): error C2989: 'v8::Name' : class template has already been declared as a non-class template [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
          c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(86) : see declaration of 'v8::Name'
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(2072): error C2332: 'enum' : missing tag name [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
[... cut ...] ode_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(3402): error C2270: 'ByteLength' : modifiers not allowed on nonmember functions [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]
c:\documents and settings\administrator\.node-gyp\5.4.1\include\node\v8.h(3402): fatal error C1003: error count exceeds 100; stopping compilation [C:\Documents and Settings\Administrator\node_modules\node-snap7\build\node_snap7.vcxproj]

mathiask88, 2016-01-14:
Hi, you need VS 2013 or higher e.g. VS 2015 top get this working with node 4.x+ because they need C++11 support.

yasdfgr, 2016-01-14:
Ok, thank you, i'll try it - maybe you should correct the information on this website (it says, requirements VS2010 and above):
https://www.npmjs.com/package/node-snap7

mathiask88, 2016-01-14:
Yes, I should add a line that node 4.x+ needs VS13+. Will do that tomorrow. If you have VS10 and VS13/15 installed parallel it could happen that you need to pass `--msvs_version=2013/2015` to npm. Look at https://github.com/nodejs/node-gyp/issues/736 for more details.

mathiask88, 2016-01-15:
https://github.com/mathiask88/node-snap7/commit/55b12f6ae00c4fc38ea72b7fcdc2de0cea10f2ae

closed 2016-01-15 by mathiask88