mathiask88/node-snap7 #15: unable to run empty electron-starter project with example node-snap7 code

sabinin, 2016-05-09:
version of node 0.9.9 
http://prntscr.com/b1z2te - printscreen of error

please tell me what i`m doing wrong ? :)

mathiask88, 2016-05-09:
I don't have much experience with electron. Did you follow [this](https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md) guide to build the native modules?

sabinin, 2016-05-09:
Thanks for notice, will try it.

sabinin, 2016-05-17:
Hi, can you write please a version of node which snap7 does currently support ?

mathiask88, 2016-05-17:
0.8, 0.10, 0.12, 3, 4, 5 and in some days 6 also.

mathiask88, 2016-05-18:
I released a new version that supports node v6.x. Now you should be able to run newest electron starter if you `npm install` and then `electron-rebuild`.

sabinin, 2016-05-18:
Thanks a lot, will try it today!

mathiask88, 2016-05-23:
I tried it myself and it works, but I had to use `.\node_modules\.bin\electron-rebuild.cmd -f` on my windows machine to force `electron-rebuild` to rebuild the dependencies. I don't know why it skips some modules otherwise. I will close for now. If you have any questions left feel free to ask.

closed 2016-05-23 by mathiask88