mathiask88/node-snap7 #16: S7 Server Support?

j8, 2016-07-05:
Hello,

Thank you very much for the nice lib.

This is future request, but when writing tests for app using node-snap7 it will be extremely helpful to be able to mock S7 Server.

mathiask88, 2016-07-05:
Hi,

Nice to hear, that someone uses and likes the lib :grinning: Yes, that is on my list for too long. I think I can look into it in August.

mathiask88, 2016-11-18:
Hey, I got a first implementation ready. A bit late but better late than never :) If you are still interested, I can publish a new branch on sunday so you can test it.

psi-4ward, 2016-11-18:
Sadly I've currently no time to work on the project but anyways, publish 
it please! :)

mathiask88, 2016-11-20:
Unfortunately I can't make it today, but within next week for sure.

closed 2017-04-15 by mathiask88