mathiask88/node-snap7 #17: wordLength in WriteArea / ReadArea

teropes, 2016-10-25:
I'm having issues correctly understanding how wordLength parameter should behave in WriteArea / ReadArea. Not sure if this is actual issue with the library or just my misunderstanding (most likely).

**Writing DB300.DBW42 (Word) with WriteArea:**
`this.plc.WriteArea(this.plc.S7AreaDB, 300, 42, 2, this.plc.S7WLWord, wbuf, function ...)`
**Log output from snap7 dev server:**
`2016-01-25 14:25:37 [127.0.0.1] Write request, Area : DB300, Start : 42, Size : 4 --> OK`

**Writing DB300.DBDW42 (DoubleWord) with WriteArea:**
`this.plc.WriteArea(this.plc.S7AreaDB, 300, 42, 2, this.plc.S7WLDWord, wbuf, function ...)`
**Log output from snap7 dev server:**
`2016-01-25 14:25:52 [127.0.0.1] Write request, Area : DB300, Start : 42, Size : 16 --> OK`

With wordLength S7WLWord I would expect the client to write 2 bytes. However it does write 4 bytes, of which 2 bytes comes correctly from my wbuf (write buffer) and last 2 bytes are something random.

For example the "Size: 16" does not make sense to me, its not number of bits or bytes to write and not even hex or dec value of the constant S7Client.S7WLDWord

These parameters give me the result I'm expecting:
`this.plc.WriteArea(this.plc.S7AreaDB, 300, 42, 2, 2, wbuf, function ...)`
`2016-01-25 14:25:37 [127.0.0.1] Write request, Area : DB300, Start : 42, Size : 2 --> OK`

So far I have been avoiding this for Word and DoubleWord datatypes by setting my wordLength equal to byte count and it does write the correct amount without any problems. However I would like to understand what is wrong with S7WLWord not being 2 bytes but instead 4 and use the library constants correctly.

mathiask88, 2016-10-25:
The amount, in your example 2, is the account of words to write and the wordlen defines it length.  So in your first example you write 2 words with the wordlen word(2 bytes)  => 2x2 bytes. But the second example should give 2x4 bytes instead of 16. Are you sure this is correct?  The last example is correct because the parameter 2 as wordlen is S7WLByte as you can see in the docs.

teropes, 2016-10-26:
Ok got the `WriteArea `operations working as expected, swapped my incorrectly used `byteCount` variable with `wordCount` and now it works consistently with the `wordLength` constants. I had my brain stuck with idea of amount of bytes instead of words ... Thanks for the clarification, this can be considered as closed.

closed 2016-10-26 by teropes