mathiask88/node-snap7 #18: DBGet() fails with TypeError: Wrong arguments

tschombe, 2016-11-09:
First, thanks very much for that great library, i just did a few tests but already very excited to get
a node-communication to a siemens s7 working.

I want to use DBGet() function but get a TypeError: Wrong arguments when i call it:

var res = s7client.DBGet(100);

same when I use it with a callback function.

mathiask88, 2016-11-09:
Hey, I looked at the code and yes there is a bug that needs to be fixed. I never had the chance to test this function because I only own a S71200/1500. You can use the command until I fixed it if you use `var res = s7client.DBGet(100, 65535);` The second parameter should be hidden in the lib and describes the maximum size for a DB (AFAIK).

tschombe, 2016-11-09:
Thanks for your very fast response!!!
I now tried with the second parameter and it seems to work, unfortunately my sps also does not
support that function ;-)  (Got an error response #20971520 - CPU : Function not available).

So I will work with DBReads and DBWrites.

mathiask88, 2017-04-15:
fixed in https://github.com/mathiask88/node-snap7/commit/4bd69ddeb9051795c32ceb6e5225e094a3b57195

closed 2017-04-15 by mathiask88

flacohenao, 2022-06-10:
Hello, I'm using the same method DBGet(3) but the library is throwing me this:

Function not available (20971520)

wich is preatty much the same case and I do have the latest version of the library.

1.0.6

mathiask88, 2022-06-11:
Your error is from the PLC itself and not the library. Not all Siemens PLCs support all functions. See http://snap7.sourceforge.net/compatibility.html