mathiask88/node-snap7 #21: Listen to Siemens PLC

sn0opr, 2016-12-21:
Is there a way to keep listening to Siemens PLC and trigger an event when data is received?

What I want is, instead of running a ReadArea function every 100ms, I want to keep listening to Siemens PLC and wait for data that he send.

mathiask88, 2016-12-21:
I think what you are searching for is the [Partner model](http://snap7.sourceforge.net/snap7_partner.html), but this is not yet implemented in node-snap7.

sn0opr, 2016-12-22:
@mathiask88 thank you, is there any plans to implement it, and when?

mathiask88, 2016-12-23:
Yes there are plans to implement the partner but i can't tell you any date because this is a free time project. I recently added the server functionality but it is not yet released on npm.  
There should be no risk to poll the PLC because u can set the communication ressources used by the PLC cpu in TIA. I got a project where I poll the PLC nonstop and there are no issues.

sn0opr, 2016-12-23:
ok @mathiask88, thank you.

teropes, 2017-03-16:
Old thread already, but giving here my experiences with polling S7-400 and S7-1500 series. Latest project I have been working with we had 4 clients with node-snap7 connected to our S7-1500. One is "plc-poller" ticking with 500ms cycle time, reading some DB and parsing around 3500 signals out of it. Other three are running other tasks including reading and writing to plc with 1000ms cycle time.

So far I have seen no issues with the S7 CPU or our node processes lagging or failing reads / writes, it has been robust. The plc-poller process actually spends most of its cpu cycles parsing and handling the signals from the Buffer and doing other stuff with it.

mathiask88, 2017-03-17:
Cool, I'm glad to hear that node-snap7 performs well for you. Being robust is very important for this type of software and I'm happy that it is.

I will try to implement the partner as well in the coming months. And the server needs some small fixes and testing. I also got a prebuild feature near completion so that windows, mac and linux will download prebuilt binaries instead of compiling.

SherryHolmes, 2018-10-22:
> Cool, I'm glad to hear that node-snap7 performs well for you. Being robust is very important for this type of software and I'm happy that it is.
>
> I will try to implement the partner as well in the [...]

Do you support reading and writing of DB blocks for s7-1200?Test finds read and write DB block failed
#36700160 - CLI : function refused by CPU (Unknown error)

anthrogan, 2019-02-12:
@SherryHolmes do you have put/get enabled on the 1200?

mathiask88, 2019-02-12:
Please have a look at the [compatibility table](http://snap7.sourceforge.net/snap7_client.html#target_compatibility) mentioned in the [About section](https://github.com/mathiask88/node-snap7#about)