mathiask88/node-snap7 #23: Read string  from PLC

cropii, 2017-03-16:
Hello i have data type string[6] (offset 2 to 10) i want to get value of it in node-snap7 but i noob and can't get it can you please help me.

teropes, 2017-03-16:
Hi,

Read number of bytes with S7Client.DBRead(dbNumber, start, size, [callback]). Callback result is node.js Buffer. You can output string representation of the buffer with result.toString('utf-8').

cropii, 2017-03-16:
Man, big thank you !
i was trying to unpack it.
Maybe you can help me write string to PLC ? it's doing backward?

cropii, 2017-03-16:
Also maybe could you please help me  i trying to write data i use objpack to pack buffer:

datatype is "real", db is 1, offset 16 to 20;
var paste_buffer = objpack.pack('f>', 2.2);
	
	s7client.WriteArea(0x84, 1, 16, 4, 0x08, paste_buffer, function(err) {
		if(err) return console.log(' Working_con. Code #' + err + ' - ' + s7client.ErrorText(err));
	});

then i read value from this offset and get NaN

closed 2017-03-16 by cropii