mathiask88/node-snap7 #24: GetAgBlockInfo() not working

Yaro96, 2017-03-23:
I tried to use this function to get the DB size on a Siemens s7-300, using the callback it doesn't enter into it at all. Without the callback it just returns "false" with no errors.
[new 2.txt](https://github.com/mathiask88/node-snap7/files/865039/new.2.txt)

mathiask88, 2017-03-23:
The function is working for me. As a side note: Returning in the callback function (`return res.MC7Size;`) will not return the value for `getDBLength()` 😉

Yaro96, 2017-03-23:
In my case, it doesn't enter in the callback at all, it doesn't even print "I'm here!"

mathiask88, 2017-03-23:
Are you sure that the client gets connected at all? AFAIK 300er are on slot 1 and not 2.

Yaro96, 2017-03-23:
Yes, I'm sure it works perfectly with other functions and the Snap7 demo client.
It doesn't pop any connection errors and if I put a console.log just before the GetAgBlockInfo(), it prints.

mathiask88, 2017-03-23:
If I can't reproduce your error I don't know how I can help you. Can you try your script with the snap7 demo server? And maybe post your node.js and node-snap7 versions?

Yaro96, 2017-03-28:
I figured it out, it was an asyncrous request problem. Sorry for bothering you.

closed 2017-03-28 by Yaro96

mathiask88, 2017-03-28:
Good to hear, that you found it. No problem :)