mathiask88/node-snap7 #28: s7client ErrorText function

fabioriva, 2017-05-05:
Just to inform you that after upgrading to npm 0.3.0 my working application was exiting with error when calling S7Client.ErrorText(err) function.
Then i upgraded to npm 0.3.1 and now it's returning correctly the error string.
...
By the way, thank you Mathias for your work, i am using node-snap7 in production together with electronjs + reactjs.
Tested succesfully  node-snap7 with simatic S7-300, S7-1500 and also VIPA CPU.

mathiask88, 2017-05-05:
Yeah, the error functions in snap7 1.4.2 have a stack buffer overflow bug. Unfortunately I didn't know that issue and it slipped through in 0.3.0. I monkey patched the upstream snap7 to fix this issue. Sorry for the broken release :(

Nice to hear that someone uses the library and that it is robust enough for production :) 

If privacy allows I would like to see a screenshot of your electron app.

fabioriva, 2017-05-05:
The app is used to manage a full automatic parking system. It is reading and writing data to 5 differents simatic s7-300 CPU. Following some screenshots of the application:

System overview
[image]
Parking occupancy
[image]
PLC I/O status (s7 read data polling every 250msec)
[image]

closed 2017-05-05 by fabioriva

mathiask88, 2017-05-05:
Thanks for the sneak peek. Sometimes I think not many people are really using this lib, but this is awesome!! :)