mathiask88/node-snap7 #29: Unable to run electron app that using node-snap7

sabinin, 2017-07-05:
Hi, i was trying to create an simple electron application based on node-snap7 pakage, it works fine under Linux but when i create application for Windows it do not run (when i exclude strings with declaration of variables required for node-snap7 it works on windows too ), can you help to solve this please?

mathiask88, 2017-07-05:
I can try to. What do you mean with 
  > when i exclude strings with declaration of variables required for node-snap7 it works on windows too

Some code would help I think.

sabinin, 2017-07-06:
var snap7 = require('node-snap7');
var s7client = new snap7.S7Client();

I mean when i just remove this two strings where we declare variables for node-snap7 - then application works on windows. "ERROR_INSTALL_PACKAGE_INVALID 1620 This installation package could not be opened. Contact the application vendor to verify that this is a valid Windows Installer package." - this is the error that windows throw when i`m trying to launch application, https://drive.google.com/open?id=0B3ok7GAAR880V2ttRGVUcXNVbnM - here you can find archive with application.

To buid application: 
npm install
./node_modules/.bin/electron-rebuild

command to build app:
electron-packager . --overwrite --asar=true --platform=win32 --arch=x64 --icon=icon.ico
To run built aplication you should paste this file to application folder https://drive.google.com/open?id=0B3ok7GAAR880RE4wempBa05VZlk

mathiask88, 2017-07-06:
It is working for me. Are you building the package for windows on linux or windows?

sabinin, 2017-07-06:
On linux for windows.

Noctua36, 2017-07-06:
@sabinin , did you have Wine (+1.8) and Mono (+4.2) installed? (see: [Multi Platform Build](https://github.com/electron-userland/electron-builder/wiki/Multi-Platform-Build))

sabinin, 2017-07-06:
Just Wine, thanks, i`ll try with Mono.

mathiask88, 2017-07-06:
The article posted by @Noctua36 states that cross building is not really possible if you use native packages in your app. 

>If your app has native dependencies, it can be compiled only on the target platform. prebuild is a solution, but most node modules don't provide prebuilt binaries.

I want to implement `prebuild` for `node-snap7` but I had no time yet.

The easiest solution would be a build server or just a VM.

sabinin, 2017-07-06:
Thanks, you right, i built it under windows and now it works.

closed 2017-07-07 by mathiask88

mathiask88, 2017-07-08:
I implemented prebuilts with `0.4.0` just to let you know. Maybe that helps for cross packing electron apps.