mathiask88/node-snap7 #31: Issue reading and writing mulitple bits

acrippa, 2017-11-21:
Hello,
I'm having problems writing and reading multiple bits with one call of "ReadArea".

`s7client.ReadArea(s7client.S7AreaDB, 1, 0, 2, s7client.S7WLBit, function(err,res){
          if(err)
              return console.log(' >> DBRead failed. Code #' + err + ' - ' + s7client.ErrorText(err));
        console.log(" >> DBRead succesful",res)
        });`
I get the error  ">> DBRead failed. Code #10485760 - CPU : Invalid Transport size".
Tested with a CPU S7-314C-2 PN/DP and a CPU S7-1511-1 PN.

I get the same error trying to write multiple bits using WriteArea.
Everything goes fine if I read(write) those bits one by one with multiple calls of "ReadArea"("WriteArea").

I have no problems reading and writing multiple bytes or words.

What could be the cause of that error?

mathiask88, 2017-11-21:
Hey,
as far as I remeber the S7 Protocol does not allow reading more than one bit at a time. If you want to do this you have to read the whole byte and extract your bits.

mathiask88, 2017-12-19:
Another option would be to use the [MultiVars ](https://github.com/mathiask88/node-snap7/blob/master/doc/client.md#read-multi-vars) functions. But I think you got a solution for your issue, so I'll close this. If you still got problems, feel free to reopen.

closed 2017-12-19 by mathiask88