mathiask88/node-snap7 #32: server in docker

chtbof, 2017-11-29:
How use server function in docker. It run ok in docker but can't be connected. 
s7server.StartTo('127.0.0.1');
docker run -p 102:102 
not work

mathiask88, 2017-11-29:
I didn't work with docker yet, but maybe this repo will help you https://github.com/Noctua36/docker-node-snap7

Noctua36, 2017-11-29:
To bind with port numbers less than 1024, you have to run the container in privileged mode (`docker run -p102:102 --privileged`), or, recommended, just add this capability to it (`docker run -p102:102 --cap-add=NET_BIND_SERVICE`)

mathiask88, 2017-11-30:
@Noctua36 Thanks for the help :) If the port is the problem, then there are also the possibilities to change the server port with `S7Server.SetParam()` or if you run a kernel >4.11 you can set `--sysctl net.ipv4.ip_unprivileged_port_start=0` as a docker parameter to disable privileged ports.

mathiask88, 2017-12-19:
@chtbof Did this solve your issue?

chtbof, 2017-12-20:
Not really. But using as server is just for testing. I am using node-snap7 for record datalog to sql server. The node-snap7 working great.

psi-4ward, 2018-01-18:
I start my dev-server usually on a Port > 1024 to avoid bind restrictions in userspace

```javascript
const s7server = new snap7.S7Server();
s7server.SetParam(s7server.LocalPort, 9102);
```

closed 2018-07-08 by mathiask88