mathiask88/node-snap7 #33: How to get Merker block size

BrkDalkiran, 2018-01-03:
Hi there,
Thank you for your great package ...
We can get dbblocksize with 'GetAgBlockInfo' method and 'MC7Size' field. 
We need to read Merker data from PLC and plan to use MBRead method. 
But we don't want read single address and we want full block.
So we need MerkerBlock size. 
Is there a method or property for this requirement. 
Thank you.

mathiask88, 2018-01-07:
I don't know if I understand your question right, but AFAIK the Merker blocks are a hardware dependent configuration. For example a S7-312 has Merkerbits from 0.0 to 255.7, Merkerbytes from 0 to 255, Merkerwords from 0 to 254 and Merkerdoublewords from 0 to 252.

BrkDalkiran, 2018-07-04:
I use the GetCpuInfo for hardware check. 
It's work on S7-3XX and S7-4XX, 'ModuleTypeName' is enough for check. 
We set the 'Amount' value accordingly. Added a new PLC on system. This is a S7-1200.

GetCpuInfo is not working on this.
Error num: 12582912, error text: 'Item not available'
How else can we control it?

mathiask88, 2018-07-04:
Hey, Siemens has a new protocol for their new 1200,1500. They still support the legacy protocol, but not for all functions. That's the reason for the compatibility table. If the CPUs have these fixed sizes you could try to read MB4097 and if that errors it is a 4096er and if it works it's probably a 8192er. Or hardcode it, or store it in a DB . What's the reason that you want to read this from the CPU and don't hard code it like `192.x.x.x has 4096 merker bytes`?

closed 2018-11-12 by mathiask88