mathiask88/node-snap7 #34: node-Crash on MacOS when node-snap7 is used on nodejs 4!

Apollon77, 2018-01-09:
Hi,

we use node-snap7 in a project and in our Travis-CI tests on MacOS the node process crashed immediately when starting with:

Assertion failed: (object->InternalFieldCount() > 0), function Wrap, file ../node_modules/nan/nan_object_wrap.h, line 55.

https://travis-ci.org/ioBroker/ioBroker.s7/jobs/326705863#L456
I tried also "build from source", same happens there.

Any idea what causes this?

Apollon77, 2018-01-09:
I think the crash happens in https://github.com/ioBroker/ioBroker.s7/blob/master/test/lib/s7Server.js

Apollon77, 2018-01-09:
On Windows for node 4 (on Appveyor) it just hangs (and Appveyor kills the test after 1h) basically at the same place in code where it brings that error on MacOS (I assume)

Apollon77, 2018-01-09:
Need to correct myself again: also happens on Linux. So irbid nodejs 4 on all platforms where this problem exists.

Apollon77, 2018-01-09:
On Linux the error is:

node: ../node_modules/nan/nan_object_wrap.h:55: void Nan::ObjectWrap::Wrap(v8::Local<v8::Object>): Assertion `object->InternalFieldCount() > 0' failed.

Apollon77, 2018-01-09:
Additonal Link to the Travis-ci/Appveyor tests for my project where you can see this:

Linux: https://travis-ci.org/ioBroker/ioBroker.s7/jobs/326887620#L840
MacOS: https://travis-ci.org/ioBroker/ioBroker.s7/jobs/326887626#L452
Windows: https://ci.appveyor.com/project/Apollon77/iobroker-s7/build/test-65/job/egcig2l9pa98cd63#L266 ( you only see that there is no output anymore and Appveyor killed it after 1h

mathiask88, 2018-01-10:
@Apollon77 I'm currently searching for the reason, but I don't have any idea yet :confused:

Apollon77, 2018-01-10:
Thank you in any case for your efforts! Hope you find the "holy gral" soon :-) I'm no C/C++ guy else I would offer help.
But you can reproduce it?

mathiask88, 2018-01-11:
Yes I could, thanks for reporting. If I google that error I see several issues but no real solution. I will ask in nodejs/help if I can't find a cause.

mathiask88, 2018-01-19:
Should be fixed by e0ef6406b9aa4666946ab788f5e6d2f7324a6bdd. I will release a new version this weekend.

Apollon77, 2018-01-19:
Cool!
As soon as it is released I will update the dep in my project and verify it!

mathiask88, 2018-01-20:
@Apollon77 You can try `1.0.0` now :)

Apollon77, 2018-01-20:
Verified. Tests show green!! Thank you very much

closed 2018-01-20 by Apollon77