mathiask88/node-snap7 #35: Module version mismatch

hillcz, 2018-01-26:
Hi 
If i try SNAP7 on LTS node 6.10.2
I have err. message  Module version mismatch. Expected 47, got 48.
Can you fix it ? 
THX

mathiask88, 2018-01-26:
Did you change your node version or is this a fresh installation? If you change major node versions you always have to `npm rebuild `.

Noctua36, 2018-01-26:
Are you using it with electron? If so, you can install the `electron-rebuild` package and add this script on package.json : `"postinstall": "electron-rebuild"`

hillcz, 2018-01-27:
Nodejs was preinstaled by myscada app. node-snap7 I am installing by nmp install

mathiask88, 2018-01-31:
@hillcz If I can't reproduce this, I can only guess what your issue is. I think it is not an issue with `node-snap7` itself but with your environment you are installing. Did you solve it?

hillcz, 2018-02-01:
Yes I fixed it. I dont know why, but NTP VERSION return me NODE 6.10.2 and i have instaled 5.4.1.
Now it is working. Thanks for perfect module. Sorry for fake issue and my horible english .

closed 2018-02-01 by hillcz