mathiask88/node-snap7 #36: Memory issue

benoitblin, 2018-02-19:
Thanks for this great wrapper. I use the client to read periodically DBs of PLCs in a loop. Unfortunately, I always have a memory issue after a random time (Ubuntu 16.04 platform, Node.JS v8.9.0), like :
- Error in `node': double free or corruption (out): 0x00000000039d6a20
or :
- Error in `node': free(): invalid pointer: 0x0000000002e938a0

May it be a bug ? Or bad design of my app ?

Here an example of one of this trace :
*** Error in `node': free(): invalid pointer: 0x0000000002e938a0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fb649f587e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fb649f6137a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fb649f6553c]
/path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node(_ZN10node_snap78IOWorker16HandleOKCallbackEv+0x398)[0x7fb645a8b508] /path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node(_ZN3Nan11AsyncWorker12WorkCompleteEv+0x83)[0x7fb645a8f8f3]
/path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node(_ZN3Nan20AsyncExecuteCompleteEP9uv_work_s+0x15)[0x7fb645a8f7a5] 
node[0x14549d5]
node[0x1456a8b]
node[0x14685e8]
node(uv_run+0x156)[0x1457416]
node(_ZN4node5StartEP9uv_loopsiPKPKciS5+0xc75)[0x122a035]
node(_ZN4node5StartEiPPc+0x163)[0x1222c93]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fb649f01830]
node[0x8ae4c1]

======= Memory map: ========
00400000-01fa2000 r-xp 00000000 08:01 402502 /home/me/.nvm/versions/node/v8.9.0/bin/node
021a2000-021bd000 rw-p 01ba2000 08:01 402502 /home/me/.nvm/versions/node/v8.9.0/bin/node
021bd000-021d4000 rw-p 00000000 00:00 0 02c72000-034a4000 rw-p 00000000 00:00 0 [heap]
[...]
7fb645a62000-7fb645ac0000 r-xp 00000000 08:01 1072693 /path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node
7fb645ac0000-7fb645cbf000 ---p 0005e000 08:01 1072693 /path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node
7fb645cbf000-7fb645cc1000 r--p 0005d000 08:01 1072693 /path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node
7fb645cc1000-7fb645cc7000 rw-p 0005f000 08:01 1072693 /path/to/my/app/node_modules/node-snap7/build/Release/node_snap7.node
7fb645cc7000-7fb645cd2000 r-xp 00000000 08:01 786995 /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fb645cd2000-7fb645ed1000 ---p 0000b000 08:01 786995 /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fb645ed1000-7fb645ed2000 r--p 0000a000 08:01 786995 /lib/x86_64-linux-gnu/libnss_files-2.23.so
7fb645ed2000-7fb645ed3000 rw-p 0000b000 08:01 786995 /lib/x86_64-linux-gnu/libnss_files-2.23.so
[...]
7fb649ee1000-7fb64a0a1000 r-xp 00000000 08:01 786931 /lib/x86_64-linux-gnu/libc-2.23.so
7fb64a0a1000-7fb64a2a1000 ---p 001c0000 08:01 786931 /lib/x86_64-linux-gnu/libc-2.23.so
7fb64a2a1000-7fb64a2a5000 r--p 001c0000 08:01 786931 /lib/x86_64-linux-gnu/libc-2.23.so
7fb64a2a5000-7fb64a2a7000 rw-p 001c4000 08:01 786931 /lib/x86_64-linux-gnu/libc-2.23.so
7fb64a2a7000-7fb64a2ab000 rw-p 00000000 00:00 0
7fb64a2ab000-7fb64a2c3000 r-xp 00000000 08:01 787018 /lib/x86_64-linux-gnu/libpthread-2.23.so
7fb64a2c3000-7fb64a4c2000 ---p 00018000 08:01 787018 /lib/x86_64-linux-gnu/libpthread-2.23.so
7fb64a4c2000-7fb64a4c3000 r--p 00017000 08:01 787018 /lib/x86_64-linux-gnu/libpthread-2.23.so
7fb64a4c3000-7fb64a4c4000 rw-p 00018000 08:01 787018 /lib/x86_64-linux-gnu/libpthread-2.23.so
7fb64a4c4000-7fb64a4c8000 rw-p 00000000 00:00 0
7fb64a4c8000-7fb64a4de000 r-xp 00000000 08:01 786957 /lib/x86_64-linux-gnu/libgcc_s.so.1
7fb64a4de000-7fb64a6dd000 ---p 00016000 08:01 786957 /lib/x86_64-linux-gnu/libgcc_s.so.1
7fb64a6dd000-7fb64a6de000 rw-p 00015000 08:01 786957 /lib/x86_64-linux-gnu/libgcc_s.so.1
7fb64a6de000-7fb64a7e6000 r-xp 00000000 08:01 786972 /lib/x86_64-linux-gnu/libm-2.23.so
7fb64a7e6000-7fb64a9e5000 ---p 00108000 08:01 786972 /lib/x86_64-linux-gnu/libm-2.23.so
7fb64a9e5000-7fb64a9e6000 r--p 00107000 08:01 786972 /lib/x86_64-linux-gnu/libm-2.23.so
7fb64a9e6000-7fb64a9e7000 rw-p 00108000 08:01 786972 /lib/x86_64-linux-gnu/libm-2.23.so
7fb64a9e7000-7fb64ab59000 r-xp 00000000 08:01 921735 /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fb64ab59000-7fb64ad59000 ---p 00172000 08:01 921735 /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fb64ad59000-7fb64ad63000 r--p 00172000 08:01 921735 /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fb64ad63000-7fb64ad65000 rw-p 0017c000 08:01 921735 /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7fb64ad65000-7fb64ad69000 rw-p 00000000 00:00 0
7fb64ad69000-7fb64ad70000 r-xp 00000000 08:01 787024 /lib/x86_64-linux-gnu/librt-2.23.so
7fb64ad70000-7fb64af6f000 ---p 00007000 08:01 787024 /lib/x86_64-linux-gnu/librt-2.23.so
7fb64af6f000-7fb64af70000 r--p 00006000 08:01 787024 /lib/x86_64-linux-gnu/librt-2.23.so
7fb64af70000-7fb64af71000 rw-p 00007000 08:01 787024 /lib/x86_64-linux-gnu/librt-2.23.so
7fb64af71000-7fb64af74000 r-xp 00000000 08:01 786945 /lib/x86_64-linux-gnu/libdl-2.23.so
7fb64af74000-7fb64b173000 ---p 00003000 08:01 786945 /lib/x86_64-linux-gnu/libdl-2.23.so
7fb64b173000-7fb64b174000 r--p 00002000 08:01 786945 /lib/x86_64-linux-gnu/libdl-2.23.so
7fb64b174000-7fb64b175000 rw-p 00003000 08:01 786945 /lib/x86_64-linux-gnu/libdl-2.23.so
7fb64b175000-7fb64b19b000 r-xp 00000000 08:01 786907 /lib/x86_64-linux-gnu/ld-2.23.so
7fb64b39a000-7fb64b39b000 r--p 00025000 08:01 786907 /lib/x86_64-linux-gnu/ld-2.23.so
7fb64b39b000-7fb64b39c000 rw-p 00026000 08:01 786907 /lib/x86_64-linux-gnu/ld-2.23.so
7fb64b39c000-7fb64b39d000 rw-p 00000000 00:00 0
7ffd437d6000-7ffd437f7000 rw-p 00000000 00:00 0 [stack]
7ffd437fb000-7ffd437fd000 r--p 00000000 00:00 0 [vvar]
7ffd437fd000-7ffd437ff000 r-xp 00000000 00:00 0 [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0 [vsyscall]

mathiask88, 2018-02-19:
Could be an issue in `node-snap7` of course, but the code is kinda old and this is the first time someone got this error. Let's try to find out.. A minimal example to reproduce would be great. What function are you calling `DBGet` or `ReadArea/ReadDB`? Could you build `node-snap7` from source with `--debug` configuration to get more infos like the actual LOC?

benoitblin, 2018-02-20:
Here is an example of my code. All 2s, I read all DBs of six 1511.
Everything is OK, but after a random time (5 minutes, 1 hour, 5 hours, ...), this code crash with error below.
I'll try to build node-snap7 with --debug.

```
const snap7 = `require('node-snap7');`

class Client
{
    constructor(host_ip)
    {
        this.s7client = new snap7.S7Client();
        this.host_ip = host_ip;
        this.connect();
    }

    /**
     * Connect to S7 server
     * @param integer time before reconnect (in ms)
     */
    connect(reconnect = 10000)
    {
        if(true == this.s7client.Connected() || null == this.host_ip) {
            return;
        }
        const self = this;
        this.s7client.ConnectTo(this.host_ip, 0, 0, function(err) {
            if(err) {
                console.error("[S7] Connection error on " + self.host_ip + " : " + self.s7client.ErrorText(err));
                setTimeout(() => self.connect(reconnect), reconnect);
            } else {
                console.log(' [*] S7 : connected to ' + self.host_ip);
            }
        });
    }

    /**
     * Update DB from S7 server
     * @param DB
     */
    update(db)
    {
        if(false == this.s7client.Connected()) {
            return;
        }
        const self = this;
        this.s7client.DBRead(db.num, 0, db.length, function (err, res) {
            if(err) {
                return;
            }
            db.buffer = res;
        });
    }
}

class DB
{
    /**
     * @param int n° of DB
     * @param int length of DB 
     */
    constructor(num, length)
    {
        this.num = num;
        this.length = length;
	this.buffer = null;
    }
    	
    /**
     * Get a part of buffer
     * @return Buffer
     */
    getBuffer(offset, length)
    {
        return this.buffer.slice(offset, offset + length);
    }
}

// Init DB objects
const dbs = new Map();
dbs.set(1, [ 
  new DB(6, 154), 
  new DB(16, 98), 
  new DB(1001, 34), 
  new DB(1003, 20000), 
  new DB(1004, 10000), 
  new DB(1005, 448), 
  new DB(1006, 327), 
  new DB(1030, 148), 
  new DB(1031, 12), 
  new DB(1500, 12), 
  new DB(1600, 1), 
  new DB(1606, 10), 
  new DB(1650, 26)
]);
dbs.set(2, [ 
  new DB(6, 154), 
  new DB(16, 98), 
  new DB(1001, 34), 
  new DB(1003, 20000), 
  new DB(1004, 10000), 
  new DB(1005, 448), 
  new DB(1006, 327), 
  new DB(1030, 148), 
  new DB(1031, 12), 
  new DB(1500, 12), 
  new DB(1600, 1), 
  new DB(1606, 10), 
  new DB(1650, 26)
]);
dbs.set(3, [ 
  new DB(6, 154), 
  new DB(16, 98), 
  new DB(1001, 34), 
  new DB(1003, 20000), 
  new DB(1004, 10000), 
  new DB(1005, 448), 
  new DB(1006, 327), 
  new DB(1030, 148), 
  new DB(1031, 12), 
  new DB(1500, 12), 
  new DB(1600, 1), 
  new DB(1606, 10), 
  new DB(1650, 26)
]);
dbs.set(4, [ 
  new DB(6, 154), 
  new DB(16, 98), 
  new DB(1001, 34), 
  new DB(1003, 20000), 
  new DB(1004, 10000), 
  new DB(1005, 448), 
  new DB(1006, 327), 
  new DB(1030, 148), 
  new DB(1031, 12), 
  new DB(1500, 12), 
  new DB(1600, 1), 
  new DB(1606, 10), 
  new DB(1650, 26)
]);
dbs.set(5, [ 
  new DB(6, 154), 
  new DB(16, 98), 
  new DB(1001, 34), 
  new DB(1003, 20000), 
  new DB(1004, 10000), 
  new DB(1005, 448), 
  new DB(1006, 327), 
  new DB(1030, 148), 
  new DB(1031, 12), 
  new DB(1500, 12), 
  new DB(1600, 1), 
  new DB(1606, 10), 
  new DB(1650, 26)
]);
dbs.set(6, [ 
  new DB(6, 154), 
  new DB(16, 98), 
  new DB(1001, 34), 
  new DB(1003, 20000), 
  new DB(1004, 10000), 
  new DB(1005, 448), 
  new DB(1006, 327), 
  new DB(1030, 148), 
  new DB(1031, 12), 
  new DB(1500, 12), 
  new DB(1600, 1), 
  new DB(1606, 10), 
  new DB(1650, 26)
]);

// Init clients
const clients = {
  1: new Client('192.168.0.11'),
  2: new Client('192.168.0.12'),
  3: new Client('192.168.0.13'),
  4: new Client('192.168.0.14'),
  5: new Client('192.168.0.15'),
  6: new Client('192.168.0.16')
];

// Loop on reading operation
function run() {
  
    for(let i in clients) {
        for(let db of dbs.get(i)) {
            clients[i].update(db);
        }
    }

    setTimeout(run, 2000);
}

run();
```

benoitblin, 2018-02-21:
I've built node-snap7 from source with --debug configuration, but I don't know how to get more infos than this backtrace.

*** Error in `node': corrupted size vs. prev_size: 0x0000000002d6bc00 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f513c6ca7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x80dfb)[0x7f513c6d3dfb]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f513c6d753c]
/path/to/my/app/node_modules/node-snap7/build/Debug/node_snap7.node(_ZN10node_snap78IOWorkerD0Ev+0x24)[0x7f513a4025ae]
/path/to/my/app/node_modules/node-snap7/build/Debug/node_snap7.node(_ZN3Nan11AsyncWorker7DestroyEv+0x2a)[0x7f513a40194a]
/path/to/my/app/node_modules/node-snap7/build/Debug/node_snap7.node(_ZN3Nan20AsyncExecuteCompleteEP9uv_work_s+0x45)[0x7f513a401ae1]
node[0x14549d5]
node[0x1456a8b]
node[0x14685e8]
node(uv_run+0x156)[0x1457416]
node(_ZN4node5StartEP9uv_loop_siPKPKciS5_+0xc75)[0x122a035]
node(_ZN4node5StartEiPPc+0x163)[0x1222c93]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f513c673830]
node[0x8ae4c1]

mathiask88, 2018-04-13:
Is this still a thing? I got some spare time now and I can't repro this on my machine so I'll try to set up a VM for this. Sorry for the delay :/

benoitblin, 2018-04-16:
Thank you for attention. Yes I still have this error. Let me know if I can help you.

coojames1010, 2019-04-02:
I have written a program that is also running into this problem after random amounts of time. I am attaching my error log below. What else can I do to help to pin point what is causing the issue here?

`*** Error in `node /home/CommandService': free(): invalid pointer: 0x0000000003f5a4d0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81489)[0x7f9793344489]
/home/CommandService/my_modules/plc-siemens/node_modules/node-snap7/build/Release/node_snap7.node(_ZN10node_snap78IOWorkerD0Ev+0xb7)[0x7f9790886207]
node /home/CommandService[0xa45565]
node /home/CommandService[0xa498df]
node /home/CommandService[0xa5a5c8]
node /home/CommandService(uv_run+0x14b)[0xa4a21b]
node /home/CommandService(_ZN4node5StartEPN2v87IsolateEPNS_11IsolateDataERKSt6vectorISsSaISsEES9_+0x565)[0x8e6f45]
node /home/CommandService(_ZN4node5StartEiPPc+0x469)[0x8e5239]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f97932e53d5]
node /home/CommandService[0x89ed85]
======= Memory map: ========
00400000-02296000 r-xp 00000000 fd:00 137437568                          /usr/bin/node
02495000-02497000 r--p 01e95000 fd:00 137437568                          /usr/bin/node
02497000-024ac000 rw-p 01e97000 fd:00 137437568                          /usr/bin/node
024ac000-024c6000 rw-p 00000000 00:00 0
03ef8000-043c6000 rw-p 00000000 00:00 0                                  [heap]
5565300000-5565380000 rw-p 00000000 00:00 0
f9b8b80000-f9b8c00000 rw-p 00000000 00:00 0
13f9cb00000-13f9cb80000 rw-p 00000000 00:00 0
4444bb80000-4444bc00000 rw-p 00000000 00:00 0
5749fa80000-5749fb00000 rw-p 00000000 00:00 0
59aaa180000-59aaa200000 rw-p 00000000 00:00 0
64c5bc00000-64c5bc80000 rw-p 00000000 00:00 0
6731a700000-6731a703000 rw-p 00000000 00:00 0
7315fc00000-7315fc83000 rw-p 00000000 00:00 0
917b1480000-917b1500000 rw-p 00000000 00:00 0
96ebcf80000-96ebd000000 rw-p 00000000 00:00 0
a16e8a80000-a16e8b00000 rw-p 00000000 00:00 0
a2032a00000-a2032a80000 rw-p 00000000 00:00 0
a68cf100000-a68cf180000 rw-p 00000000 00:00 0
b3bb3c00000-b3bb3c80000 rw-p 00000000 00:00 0
b4a67b00000-b4a67b80000 rw-p 00000000 00:00 0
b8f56680000-b8f56700000 rw-p 00000000 00:00 0
bdd23a80000-bdd23b00000 rw-p 00000000 00:00 0
c8c9fd00000-c8c9fd80000 rw-p 00000000 00:00 0
d9f98580000-d9f98600000 rw-p 00000000 00:00 0
e5586280000-e5586300000 rw-p 00000000 00:00 0
ed26ae00000-ed26ae80000 rw-p 00000000 00:00 0
10edaee80000-10edaef00000 rw-p 00000000 00:00 0
112d47000000-112d47080000 rw-p 00000000 00:00 0
112f45080000-112f45100000 rw-p 00000000 00:00 0
113b48e00000-113b48e80000 rw-p 00000000 00:00 0
12221ea00000-12221ea03000 rw-p 00000000 00:00 0
12221ea03000-12221ea80000 r--p 00000000 00:00 0
122b65900000-122b65980000 rw-p 00000000 00:00 0
1298a4580000-1298a4600000 rw-p 00000000 00:00 0
13e531600000-13e531680000 rw-p 00000000 00:00 0
13fe66700000-13fe66780000 rw-p 00000000 00:00 0
149967c80000-149967d00000 rw-p 00000000 00:00 0
152870680000-152870700000 rw-p 00000000 00:00 0
1528d3f80000-1528d4000000 rw-p 00000000 00:00 0
1557e2f80000-1557e3000000 rw-p 00000000 00:00 0
15d70d680000-15d70d700000 rw-p 00000000 00:00 0
163334100000-163334180000 rw-p 00000000 00:00 0
16a71ff80000-16a720000000 rw-p 00000000 00:00 0
16bbcbc00000-16bbcbc80000 rw-p 00000000 00:00 0
17265be00000-17265be80000 rw-p 00000000 00:00 0
17b58f000000-17b58f080000 rw-p 00000000 00:00 0
1825d6980000-1825d6a00000 rw-p 00000000 00:00 0
183ded5bc000-183ded5c4000 rw-p 00000000 00:00 0
189832780000-189832800000 rw-p 00000000 00:00 0
1a67ea281000-1a67ea300000 ---p 00000000 00:00 0
1a67ea300000-1a67ea303000 rw-p 00000000 00:00 0
1a67ea303000-1a67ea304000 ---p 00000000 00:00 0
1a67ea304000-1a67ea37f000 r-xp 00000000 00:00 0
1a67ea37f000-1a67ea380000 ---p 00000000 00:00 0
[... 128 lines cut ...]
7f9793dc3000-7f9793dc4000 rw-p 00101000 fd:00 201327990                  /usr/lib64/libm-2.17.so
7f9793dc4000-7f9793ead000 r-xp 00000000 fd:00 201331475                  /usr/lib64/libstdc++.so.6.0.19
7f9793ead000-7f97940ac000 ---p 000e9000 fd:00 201331475                  /usr/lib64/libstdc++.so.6.0.19
7f97940ac000-7f97940b4000 r--p 000e8000 fd:00 201331475                  /usr/lib64/libstdc++.so.6.0.19
7f97940b4000-7f97940b6000 rw-p 000f0000 fd:00 201331475                  /usr/lib64/libstdc++.so.6.0.19
7f97940b6000-7f97940cb000 rw-p 00000000 00:00 0
7f97940cb000-7f97940d2000 r-xp 00000000 fd:00 203032083                  /usr/lib64/librt-2.17.so
7f97940d2000-7f97942d1000 ---p 00007000 fd:00 203032083                  /usr/lib64/librt-2.17.so
7f97942d1000-7f97942d2000 r--p 00006000 fd:00 203032083                  /usr/lib64/librt-2.17.so
7f97942d2000-7f97942d3000 rw-p 00007000 fd:00 203032083                  /usr/lib64/librt-2.17.so
7f97942d3000-7f97942d5000 r-xp 00000000 fd:00 201327988                  /usr/lib64/libdl-2.17.so
7f97942d5000-7f97944d5000 ---p 00002000 fd:00 201327988                  /usr/lib64/libdl-2.17.so
7f97944d5000-7f97944d6000 r--p 00002000 fd:00 201327988                  /usr/lib64/libdl-2.17.so
7f97944d6000-7f97944d7000 rw-p 00003000 fd:00 201327988                  /usr/lib64/libdl-2.17.so
7f97944d7000-7f97944f9000 r-xp 00000000 fd:00 201327621                  /usr/lib64/ld-2.17.so
7f97946e9000-7f97946f0000 rw-p 00000000 00:00 0
7f97946f1000-7f97946f2000 rw-p 00000000 00:00 0
7f97946f2000-7f97946f3000 ---p 00000000 00:00 0
7f97946f3000-7f97946f8000 rw-p 00000000 00:00 0
7f97946f8000-7f97946f9000 r--p 00021000 fd:00 201327621                  /usr/lib64/ld-2.17.so
7f97946f9000-7f97946fa000 rw-p 00022000 fd:00 201327621                  /usr/lib64/ld-2.17.so
7f97946fa000-7f97946fb000 rw-p 00000000 00:00 0
7ffe4f730000-7ffe4f751000 rw-p 00000000 00:00 0                          [stack]
7ffe4f7a7000-7ffe4f7a9000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]`

coojames1010, 2019-04-02:
After reading this thread I have also run this command:
`npm install node-snap7 --build-from-source --debug`

If I get anymore useful logs after having run that I will report back with them.

coojames1010, 2019-04-03:
After having built node-snap7 from source with the --debug flag my backtrace error looks like this:

- corrupted size vs. prev_size:

*** Error in `node /home/CommandService': corrupted size vs. prev_size: 0x0000000003ac2e20 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x80d97)[0x7f2307555d97]
/lib64/libc.so.6(+0x82295)[0x7f2307557295]
/lib64/libc.so.6(__libc_malloc+0x4c)[0x7f230755a1ac]
node /home/CommandService(_ZN4node14StreamListener13OnStreamAllocEm+0x20)[0x9a5700]
node /home/CommandService[0x9ad1ac]
node /home/CommandService[0xa547ac]
node /home/CommandService[0xa54fa0]
node /home/CommandService[0xa5a5c8]
node /home/CommandService(uv_run+0x14b)[0xa4a21b]
node /home/CommandService(_ZN4node5StartEPN2v87IsolateEPNS_11IsolateDataERKSt6vectorISsSaISsEES9_+0x565)[0x8e6f45]
node /home/CommandService(_ZN4node5StartEiPPc+0x469)[0x8e5239]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7f23074f73d5]
node /home/CommandService[0x89ed85]
======= Memory map: ========

mathiask88, 2019-04-04:
Sorry for having not much time at the moment. Unfortunately I can't reproduce this. A minimal example on a fresh VM would help, so that I am able to reproduce this error and investigate the cause. Would also be nice if you can test if this happens for all non EOL major stable node versions.

SherryHolmes, 2020-08-10:
I have the same problem.
`*** buffer overflow detected ***: node terminated
======= Backtrace: =========
/lib64/libc.so.6(__fortify_fail+0x37)[0x7f964d8416e7]
/lib64/libc.so.6(+0x116862)[0x7f964d83f862]
/lib64/libc.so.6(+0x118647)[0x7f964d841647]
/home/sherry/smydProject/plcServer/B/node_modules/_node-snap7@1.0.2@node-snap7/build/Release/node_snap7.node(_ZN10TMsgSocket10SckConnectEv+0x119)[0x7f964950bbd9]
/home/sherry/smydProject/plcServer/B/node_modules/_node-snap7@1.0.2@node-snap7/build/Release/node_snap7.node(_ZN13TIsoTcpSocket10isoConnectEv+0x48)[0x7f96494f8f88]
/home/sherry/smydProject/plcServer/B/node_modules/_node-snap7@1.0.2@node-snap7/build/Release/node_snap7.node(_ZN10TSnap7Peer11PeerConnectEv+0x19)[0x7f96494fbe59]
/home/sherry/smydProject/plcServer/B/node_modules/_node-snap7@1.0.2@node-snap7/build/Release/node_snap7.node(_ZN17TSnap7MicroClient7ConnectEv+0x1d)[0x7f964950a0ad]
/home/sherry/smydProject/plcServer/B/node_modules/_node-snap7@1.0.2@node-snap7/build/Release/node_snap7.node(_ZN10node_snap78IOWorker7ExecuteEv+0x117)[0x7f96494d9857]
node[0x15a8ec1]
/lib64/libpthread.so.0(+0x7e25)[0x7f964dafde25]
/lib64/libc.so.6(clone+0x6d)[0x7f964d827bad]
======= Memory map: ========
00400000-0205e000 r-xp 00000000 fd:02 100875731                          /home/sherry/node-v8.4.0-linux-x64/bin/node
0225d000-02279000 rw-p 01c5d000 fd:02 100875731                          /home/sherry/node-v8.4.0-linux-x64/bin/node
02279000-02291000 rw-p 00000000 00:00 0
03842000-03ee6000 rw-p 00000000 00:00 0                                  [heap]
2c05d00000-2c05d80000 ---p 00000000 00:00 0
340b900000-340b980000 ---p 00000000 00:00 0
3ab0500000-3ab0580000 ---p 00000000 00:00 0
7efd280000-7efd300000 rw-p 00000000 00:00 0
ae4a000000-ae4a080000 ---p 00000000 00:00 0
c852f00000-c852f80000 ---p 00000000 00:00 0
c905c80000-c905d5a000 rw-p 00000000 00:00 0
de3bc00000-de3bcac000 rw-p 00000000 00:00 0
216c4900000-216c4980000 ---p 00000000 00:00 0
2dec5880000-2dec5900000 ---p 00000000 00:00 0
37083b00000-37083b80000 rw-p 00000000 00:00 0
3aa40880000-3aa40900000 ---p 00000000 00:00 0`

BrkDalkiran, 2023-08-01:
hi @mathiask88,
i caught when the error occuring,

If there is an error in the code 655470, one of the following errors will definitely appear in the next db reading. I tested this 20 times
1. free(): invalid next size (fast)
2. corrupted size vs. prev_size
3. free(): invalid pointer
4. malloc(): unaligned tcache chunk detected
5. corrupted double-linked list

```
clients[deviceId]
  .DBRead(parseInt(dbNumber), 0, blocks[deviceId][dbNumber]['amount'], function (err, res) {
    if (err) {
      console.log(`DBRead error`, err, clients[deviceId].ErrorText(err))
    } else {
    }
  })
```

> DBRead error 655470  ISO : An error occurred during recv TCP : Connection timed out