mathiask88/node-snap7 #38: Type Error in WriteArea funtion

lu-kors, 2018-03-15:
I am using following code to write to DB1 a byte
```
var snap7 = require('node-snap7');
var s7client = new snap7.S7Client();
s7client.WriteArea(0x84, 1, 2, 1, 0x02, 10);
s7client.ConnectTo('192.168.1.123', 0, 1);
s7client.WriteArea(0x84, 1, 2, 1, 0x02, 10);
```
Connection can be established successfully but the following error is thrown
```
  s7client.WriteArea(0x84, 1, 2, 1, 0x02, 10);
            ^

TypeError: Wrong arguments
    at TypeError (native)
    at /var/www/zcr-www/spsconnect.js:18:13
``` 
(I already substituted s7client.S7AreaDB for debugging purpose) 
am I just using it wrong, or is there a bug in the module? 
Thanks!

mathiask88, 2018-03-15:
The last argument needs to be a node buffer.

lu-kors, 2018-03-15:
Thanks for quick response! Works finde now, may you should add it to the doc :)

closed 2018-03-15 by lu-kors