mathiask88/node-snap7 #45: 'Error: Module did not self-register.' with Worker Threads

shockingboring, 2018-11-09:
see :
[](https://github.com/nodejs/node/issues/21481)
as @mattolson mentioned 

> -- any native addons that are loaded in both the parent and worker thread (i.e. loaded more than once in the same process) need to be context aware.

mathiask88, 2018-11-12:
This addon builds on [NAN](https://github.com/nodejs/nan). As long [NAN ](https://github.com/nodejs/nan) does not support context-awareness this won't happen. Thanks for reporting!

flacohenao, 2020-04-28:
Any news on this?

mathiask88, 2020-05-12:
I am currently working on a NAPI rewrite of this addon to replace NAN. If I got the current functionality working I will have a look into this. NAPI should be capable of context-awareness. But don't expect this in the near future.