mathiask88/node-snap7 #46: S7Client.DBGet(dbNumber[, callback]) = Callback is not triggering. Async not Implemented?

Johanneslueke, 2018-11-18:
Hello,

I currently using this project for web-application I am developing. The last 3 days I was wondering why I could not read from DB till I realized I should try the synchronized way. The Blocking way of this method works just fine 👍 but the non-blocking one.... not so much

Might it be a possiblity that the async functionality is not yet implemented?

closed 2018-11-18 by mathiask88

mathiask88, 2018-11-18:
Hi,
it is implemented, but the arguments were wrong. The function check was implemented for argument number 3 and not 2. Should be fixed. You can try the current master or wait for a new release early this week (or use `DBGet(dbnumber, , callback);` But then you have to change your code again). Thanks for reporting :)

Johanneslueke, 2018-11-19:
I am happy to help. out of curiosity do you know a nice way to deliver a JSON into S7-1500 without touching every single element by hand?

mathiask88, 2018-11-19:
> Out of curiosity do you know a nice way to deliver a JSON into S7-1500 without touching every single element by hand?

What do you mean exactly? The easiest way is to write all your data in one block, but for this you need control over the DB structure in the PLC. Otherwise you can write some meta data to your JSON to describe your DB structure so that your application can process the data automatically.

commit 30a52496 (2018-11-18): fix DBGet arguments

fixes #46