mathiask88/node-snap7 #48: PLC readDB  <Pending>

anthrogan, 2019-02-12:
Any reason why a PLC wouldn't respond when its requested to readDB? In particular its the 315/7? I've checked connection resource etc and can't see anything obvious it just seems to never respond after a read attempt, The issue is obviously with the PLCs but I'm kinda lost in terms of where to look!

Johanneslueke, 2019-02-12:
If the DB is compressed reading it is not possible. Did you checked that this option is disabled?

anthrogan, 2019-02-13:
After some investigation and a chat with a siemens integrator it's appears to be down to the 343 cards in the facilities regardless of it being a 343-1 or 343 Lean. I assumed maybe a firmware difference as both cards have the same order number but there's some suggestion to the cards being physically a little different too according to their hardware version which in my mind should really change the order number:

[image]

Oh well nevermind, I will collect data via OPC through connectivity pack instead. Hopefully this is helpful to someone!

closed 2019-02-13 by anthrogan