mathiask88/node-snap7 #49: listBlocksOfType issue

chewcw, 2019-03-21:
My PLC is a ET200S (IM151-8F PN/DP) CPU.
I can successfully connect to the PLC, however when I perform ListBlocksOfType on `DB` type, I get an array of values:

```
result: [9,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,
0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
```

The total count of the array is exactly how many DB I have, and 1st number `9` is also correct, but why all subsequent number are all 0's?

Just a simple code to demonstrate:
```
client.ListBlocksOfType(0x41, (error, result) => {
	if (error) {
		console.log(`error: ${error}`);
	} else {
		console.log(`result: ${result}`);
	}
});
```

mathiask88, 2019-03-21:
Hey, looks like a bug. I'll have a look into it this weekend :)

mathiask88, 2019-03-21:
I think I found the issue. If you got some time it would be nice if you can test this [patch](https://github.com/mathiask88/node-snap7/pull/50/files) I'll then make a new release this weekend.

chewcw, 2019-03-22:
Sure I will give it a try and feedback later. Thanks for your quick action.

chewcw, 2019-03-22:
It works perfect now. Thank you so much.

closed 2019-03-22 by chewcw

mathiask88, 2019-03-22:
Thanks for the feedback, how a pair of parentheses can help :) Just published´ node-snap7@1.0.2´. If you face other issues feel free to report.