mathiask88/node-snap7 #55: Support nodejs 12

Apollon77, 2019-05-05:
Current build errors:

```
../src/node_snap7_client.cpp: In static member function ‘static void node_snap7::S7Client::Init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
../src/node_snap7_client.cpp:697:43: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
   Nan::Set(target, name, tpl->GetFunction());
                                           ^
../src/node_snap7_client.cpp:697:43: note: candidate is:
In file included from /home/travis/.node-gyp/12.1.0/include/node/node.h:63:0,
                 from ../src/node_snap7_client.h:10,
                 from ../src/node_snap7_client.cpp:6:
/home/travis/.node-gyp/12.1.0/include/node/v8.h:5947:46: note: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^
/home/travis/.node-gyp/12.1.0/include/node/v8.h:5947:46: note:   candidate expects 1 argument, 0 provided
../src/node_snap7_client.cpp: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE node_snap7::S7Client::New(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/node_snap7_client.cpp:710:53: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
     Nan::NewInstance(constructorHandle->GetFunction()).ToLocalChecked());
                                                     ^
../src/node_snap7_client.cpp:710:53: note: candidate is:
In file included from /home/travis/.node-gyp/12.1.0/include/node/node.h:63:0,
                 from ../src/node_snap7_client.h:10,
                 from ../src/node_snap7_client.cpp:6:
/home/travis/.node-gyp/12.1.0/include/node/v8.h:5947:46: note: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^
/home/travis/.node-gyp/12.1.0/include/node/v8.h:5947:46: note:   candidate expects 1 argument, 0 provided
make: *** [Release/obj.target/node_snap7/src/node_snap7_client.o] Error 1
```

mathiask88, 2019-05-05:
Hi @Apollon77 . I'll have a look in the next days. Shouldn't be hard to fix. I'm currently rewriting this module for NAPI in my rare spare-time, so in the future major version changes will be less error-prone.

Apollon77, 2019-05-06:
Thank you very much. I know that these changes are a pain ... I also have one „binary package“ Modul to maintain :-(

vweevers, 2019-05-11:
@mathiask88 FYI (because I got a build failure notification from AppVeyor for this project for some reason): node 12 is not supported yet, but there's a workaround: https://github.com/appveyor/ci/issues/2921#issuecomment-486727727

mathiask88, 2019-05-11:
@vweevers It is no real "workaround" `Update-NodeJsInstallation` did always pull a fresh version and `Install-Product` selects a preinstalled version. The preinstalled versions lag, but I have the time to wait. However thanks for being attentive! :) I don't know why you got the notifications. Maybe because `prebuild` used my appveyor account in the past.

Apollon77, 2019-05-13:
I decided for ourt perojects to switch to the "workaround method". I'm better with "latest from a certain version chain" then some old stuff, so I can tell the customers "update your node" first ...

mathiask88, 2019-06-14:
AppVeyor now has node v12 in their build images, but to be future-proof see https://github.com/mathiask88/node-snap7/commit/6738c971fc5af547f301227bbb9ce7f13e46ea27 :)

closed 2019-06-14 by mathiask88