mathiask88/node-snap7 #56: readmultivars

jandekock, 2019-05-17:
can somebody share a working readmultivars ?

this one is not working: 

var snap7 = require('node-snap7');
let res = this.client.ReadMultiVars([{"Area": snap7.S7AreaDB, "WordLen": snap7.S7WLByte,"DBNumber": 14, "Start": 46, "Amount": 1}]);

mathiask88, 2019-06-19:
@jandekock Sorry for responding that late. What is not working? What error do you get?

mariolazzari, 2019-06-25:
This one is working on my project. I do not use "" in field names.

 const vars = [
    {
      Area: s7client.S7AreaDB,
      WordLen: s7client.S7WLBit,
      DBNumber: 185,
      Start: 0,
      Amount: 1
    },
    {
      Area: s7client.S7AreaDB,
      WordLen: s7client.S7WLBit,
      DBNumber: 185,
      Start: 6,
      Amount: 1
    }
  ];

closed 2020-05-12 by mathiask88