mathiask88/node-snap7 #57: function refused by cpu

celevra, 2019-06-14:
Hi,

is it also possible to communicate with Tia Portal 1500?

regards

Celevra

Johanneslueke, 2019-06-14:
To my knowledge this is not possible because the TIA Portal only acts as a client.

celevra, 2019-06-14:
hm could be that its not really clear what i ment ;-)
programming a Siemens 1500 isn't done in S7 today but via the new TIA Portal.
I thought that it could be possible that the problems with function refused by cpu is because of that

teropes, 2019-06-14:
Yes it is possible to communicate with Siemens S7 plc's. What specific function is giving you this error?

Johanneslueke, 2019-06-14:
@celevra ok if you meant an actual s7-1500 then yes it is possible. reading writing does work fine. i haven't tried out with an R/H-systems  while those are in RunRed. But standalone does work fine because communication is done via S7-Protokoll.

mathiask88, 2019-06-14:
See https://github.com/mathiask88/node-snap7/issues/44#issuecomment-431863670

closed 2019-06-19 by mathiask88

celevra, 2019-06-19:
ok, thank you very much for your answers, i will look in my code again :-)

mathiask88, 2019-06-19:
@celevra If something is not clear, feel free to ask :)