mathiask88/node-snap7 #66: WriteArea not always working properly

BrkDalkiran, 2020-05-11:
For many reasons, we send data every 30 seconds. Only important one data. This is type S7WLBit.
We use WriteArea to send and logging. We write the data first and then read it again. Sometimes we read 1 although we want to write 0.

I made sure that the reading and writing process took about 1 second and I recorded these results. (writing and reading are done in separate applications)
Write Logs
```
May  9 19:39:22 s7[1708]: S7TagWrite: DB[300], Addr[5][S7WLBit], [0]
May  9 19:39:53 s7[1708]: S7TagWrite: DB[300], Addr[5][S7WLBit], [0]
May  9 19:40:22 s7[1708]: S7TagWrite: DB[300], Addr[5][S7WLBit], [0]
May  9 19:40:52 s7[1708]: S7TagWrite: DB[300], Addr[5][S7WLBit], [0]
May  9 19:41:22 s7[1708]: S7TagWrite: DB[300], Addr[5][S7WLBit], [0]
May  9 19:41:52 s7[1708]: S7TagWrite: DB[300], Addr[5][S7WLBit], [0]
```

- 19:40:52: 0 written
- 19:40:53: 1 readed
- 19:41:22: 0 written
- 19:41:23: 0 readed

Is there anything you can suggest for this? I can't even guess where to start

mathiask88, 2020-05-12:
Polling is totally okay for the PLC as well as node-snap7. I had a project with a S7-1500 where I continuously wrote and read data multiple times per second 24/7. With your given information I cannot help. However you write 0 to the PLC and read it after some time.. In the meantime the PLC could have changed the value. So I don't see the issue here?

BrkDalkiran, 2020-05-13:
Yes, we could not fully solve the problem. There is only one thing i am sure. Data does not change in any way when I close it writing application. 

I use the S7Client.WriteArea for writing data. Callback never returns error. For now, we will take one s7-400 and test it in the office environment for days. Faulty condition occurs at 20 PLC (s7-300, s7-400, s7-1200). As a workaround, we don't rewrite the same data. Even if there is no error in the writing process, we read it again and make sure that the data is written correctly.

When there is an improvement, I can share it here.  There is no clear information right now. I will close the issue.

closed 2020-05-13 by BrkDalkiran