mathiask88/node-snap7 #68: Install on RaspberryPi Fails

coojames1010, 2020-06-20:
Building the binaries on RaspberryPi fails. Am I doing something wrong?

`
root@raspberrypi:/home/test# sudo npm install node-snap7

> node-snap7@1.0.3 install /home/coopertech/test/node_modules/node-snap7
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.18.1 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/home/coopertech/test/node_modules/node-snap7/build'
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_msgsock.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_sysutils.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_tcpsrvr.o
../deps/snap7/src/sys/snap_tcpsrvr.cpp: In member function ‘int TCustomMsgServer::StartTo(const char*, word)’:
../deps/snap7/src/sys/snap_tcpsrvr.cpp:429:12: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 16 equals destination size [-Wstringop-truncation]
     strncpy(FLocalAddress, Address, 16);
     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_threads.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_client.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_isotcp.o
../deps/snap7/src/core/s7_isotcp.cpp: In member function ‘int TIsoTcpSocket::IsoConfirmConnection(u_char)’:
../deps/snap7/src/core/s7_isotcp.cpp:163:10: warning: variable ‘TempRef’ set but not used [-Wunused-but-set-variable]
  u_short TempRef;
          ^~~~~~~
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_partner.o
../deps/snap7/src/core/s7_partner.cpp: In member function ‘int TConnectionServer::StartTo(const char*)’:
../deps/snap7/src/core/s7_partner.cpp:252:12: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 16 equals destination size [-Wstringop-truncation]
     strncpy(FLocalAddress,Address,16);
     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_peer.o
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_server.o
../deps/snap7/src/core/s7_server.cpp: In member function ‘byte TS7Worker::WriteArea(PReqFunWriteDataItem, PReqFunWriteItem, TEv&)’:
../deps/snap7/src/core/s7_server.cpp:626:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
     if ((ReqItemPar->Area==S7AreaTM) ^ (ReqItemPar->TransportSize==S7WLTimer))
     ^~
../deps/snap7/src/core/s7_server.cpp:629:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  if ((ReqItemPar->Area==S7AreaCT) ^ (ReqItemPar->TransportSize==S7WLCounter))
  ^~
../deps/snap7/src/core/s7_server.cpp: In member function ‘void TS7Worker::SZLSystemState()’:
../deps/snap7/src/core/s7_server.cpp:1247:11: warning: ‘void* memcpy(void*, const void*, size_t)’ forming offset [5, 6] is out of the bounds [0, 4] of object ‘SZLNotAvail’ with type ‘byte [4]’ {aka ‘unsigned char [4]’} [-Warray-bounds]
     memcpy(SZL.ResData,&SZLNotAvail,sizeof(SZLSysState));
     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../deps/snap7/src/core/s7_server.cpp:27:
../deps/snap7/src/core/s7_firmware.h:36:8: note: ‘SZLNotAvail’ declared here
   byte SZLNotAvail[4] = {
        ^~~~~~~~~~~
../deps/snap7/src/core/s7_server.cpp: In member function ‘word TS7Worker::ReadArea(PResFunReadItem, PReqFunReadItem, int&, TEv&)’:
../deps/snap7/src/core/s7_server.cpp:454:42: warning: ‘BitIndex’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           if ((ByteVal & BitMask[BitIndex])!=0)
                          ~~~~~~~~~~~~~~~~^
../deps/snap7/src/core/s7_server.cpp: In member function ‘byte TS7Worker::WriteArea(PReqFunWriteDataItem, PReqFunWriteItem, TEv&)’:
../deps/snap7/src/core/s7_server.cpp:598:7: warning: ‘BitIndex’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  byte BitIndex;
       ^~~~~~~~
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_text.o
../deps/snap7/src/core/s7_text.cpp: In function ‘char* EvtSrvText(TSrvEvent&, char*, int)’:
../deps/snap7/src/core/s7_text.cpp:782:5: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
     else
     ^~~~
../deps/snap7/src/core/s7_text.cpp:785:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
  strncpy(Result, S, TextLen);
  ^~~~~~~
  CXX(target) Release/obj.target/snap7/deps/snap7/src/core/s7_micro_client.o
../deps/snap7/src/core/s7_micro_client.cpp: In member function ‘void TSnap7MicroClient::SetConnectionParams(const char*, word, word)’:
../deps/snap7/src/core/s7_micro_client.cpp:2675:13: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 16 equals destination size [-Wstringop-truncation]
      strncpy(RemoteAddress, RemAddress, 16);
      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../deps/snap7/src/core/s7_micro_client.cpp: In member function ‘int TSnap7MicroClient::SetSessionPassword(char*)’:
../deps/snap7/src/core/s7_micro_client.cpp:3306:16: warning: ‘char* strncpy(char*, const char*, size_t)’ output truncated before terminating nul copying as many bytes from a string as its length [-Wstringop-truncation]
         strncpy((char*)&opData,Password,L);
[... 34 lines cut ...]
In file included from ../src/node_snap7_server.h:12,
                 from ../src/node_snap7_server.cpp:6:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2294:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
../src/node_snap7_server.cpp: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE node_snap7::S7Server::RWBufferCallback(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/node_snap7_server.cpp:721:56: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
     if (node::Buffer::Length(info[0].As<v8::Object>()) < size) {
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
  CXX(target) Release/obj.target/node_snap7/src/snap7.o
  SOLINK_MODULE(target) Release/obj.target/node_snap7.node
  COPY Release/node_snap7.node
make: Leaving directory '/home/coopertech/test/node_modules/node-snap7/build'
npm WARN test@1.0.0 No description
npm WARN test@1.0.0 No repository field.

+ node-snap7@1.0.3
added 64 packages from 44 contributors and audited 64 packages in 34.889s

1 package is looking for funding
  run `npm fund` for details

found 0 vulnerabilities
`

mathiask88, 2020-06-20:
Actually the build does not fail. These are notes and warnings. Most of them in dependencies and probably nothing to worry about. So you should be able to use this lib.

coojames1010, 2020-06-20:
You are correct. I had another issue that was causing my code to fail. It's running perfectly now on the RaspberryPi. Thank you for the work you've put into this library!

closed 2020-06-20 by coojames1010