mathiask88/node-snap7 #74: Snap7: Exception: CLI : function refused by CPU after TIA PORTAL update

GG846, 2020-11-19:
We have an application with a Raspberry pi using SNAP7 and a S7-1214. The application was workin smoothly but the PLC was programmed in  TIA PORTAL V13.0 (wich is not compatible with new versions of TIA PORTAL). So we have proceed to update this using the following steps:
1-We've created a new proyect with the PLC in TIA PORTAL V13 and uploaded the software from the PLC
2- Upload the TIA PORTAL from V13 to V13 SP2 and then to V16. 
3-We reseted the PLC to factory settings and downloaded the software from TIA PORTAL V16.
4- We checked the I/O addresses configuration to ensure they hadn't changed. 

(this worked almost smoothly just describing the process for your better understanding) 

The rest of the system is working smoothly, but after this the raspberry pi is not working and returning: Snap7: Exception: CLI : function refused by CPU.
The main solutions given to this problems in different forums are:
-Check access level to "full" and allow GET/PUT communication check. This is propperly setted up.
-It might be a firmware incompatibility. The firmware has not change during this operation, so I guess if it was working before, it should work now.

Can you offer any other bit of information about what might be the problem here? 

Any advice will be wellcome.
Regards

mathiask88, 2020-11-19:
I don't think that I can help you much on this, because I have no TIA V16 available here. But what I know is that Siemens did some stuff regarding access rights of the data, because they implemented an OPC-UA Server in the 1200er series with the new 4.4 firmware. Maybe they are now more strict to clients that try to read/write data over the classic SIMATIC S7 ISO on TCP communication protocol.

MickaelDelanoe, 2024-12-04:
Hello,
I have the same problem.
My automatician has activated the “ALLOW PUT/GET” function.
I've asked him about the “block access optimization” property, and I'm waiting for his reply.

Have you solved your problem @GG846 ? 
Regards,

abenelfqih, 2025-05-15:
@MickaelDelanoe Hello, I am working on my last year project, I am facing the same problem, have you please solved the problem? this will help a lot

MickaelDelanoe, 2025-05-15:
Sorry @abenelfqih, but no, I haven't found the solution. This problem is still blocked for me.

mathiask88, 2025-05-16:
Try to disable Allow PUT/Get download HW info and enable again. DB optimization has to be turned of. These are the two mandatory settings and I read some threads were people were using snap7 with TIA 16 without issues.