mathiask88/node-snap7 #75: Not able to install in node15

Woiife, 2020-11-23:
Hello I reinstalled my manjaro and node-snap7 seems to stop working. I checked python version and tried a lot of things. Can you help me?
Log:

```sh: warning: setlocale: LC_ALL: cannot change locale (de)
prebuild-install WARN install No prebuilt binaries found (target=15.2.1 runtime=node arch=x64 libc= platform=linux)
sh: warning: setlocale: LC_ALL: cannot change locale (de)
/bin/sh: warning: setlocale: LC_ALL: cannot change locale (de)
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (node:events:329:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:277:12)
gyp ERR! System Linux 5.8.18-1-MANJARO
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/woife/Dokumente/VMs/node/trunk/projects/plcServer/node_modules/node-snap7
gyp ERR! node -v v15.2.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN wsplcserver@1.1.0 No description
npm WARN wsplcserver@1.1.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-snap7@1.0.5 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-snap7@1.0.5 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/woife/.npm/_logs/2020-11-23T22_13_38_628Z-debug.log```

mathiask88, 2020-11-23:
Setup your locales properly
https://github.com/nodejs/node-gyp/issues/1283

Woiife, 2020-11-24:
not only that. It seems there is a bug in a recent electron version which is causing this particular error:
https://bbs.archlinux.org/viewtopic.php?id=260980

closed 2020-11-24 by Woiife