mathiask88/node-snap7 #76: S7 1200 Poll error Last-Error Info: 36700160: CLI

lhoem, 2020-12-07:
Hallo,

habe einen Fehler bei dem IOBROKER Zugriff auf eine S7 1212 DC/DC7DC / 6ES7 212-1AE40-0XB0 / Firmware 4.3.1 / TIA Portal V16

Das normale Verbinden klappt und die Anzeige wird grün, aber wenn ich Eingangsadressen anlege (z.B. 0.0), wird sie nur noch gelb und im Log erscheint folgende Fehlermeldung:

s7.0 | 2020-12-07 08:26:38.937 | error | (3537) try reconnection: Disconnect successfull: true
-- | -- | -- | --
s7.0 | 2020-12-07 08:26:38.935 | warn | (3537) Poll error Last-Error Info: 36700160: CLI : function refused by CPU (Unknown error)
s7.0 | 2020-12-07 08:26:38.935 | warn | (3537) Poll error count: 29 code: 0x2300000 (errCliFunctionRefused)
s7.0 | 2020-12-07 08:26:38.934 | warn | (3537) EBRead error[0 - 1]: code: 0x2300000 (errCliFunctionRefused)
s7.0 | 2020-12-07 08:26:38.923 | info | (3537) Successfully connected in S7 mode
s7.0 | 2020-12-07 08:26:38.867 | info | (3537) Connect in S7 mode to 0 / 1

Kann es sein, dass es was mit der Firmware Version zu tun hat? Oder hat Jemand einen anderen Lösungsvorschlag?
Die Objekte werden auch nicht angelegt - nur der Channel

Danke Euch im Voraus!

Gruß
Lhoem

mathiask88, 2020-12-12:
See https://github.com/mathiask88/node-snap7/issues/74 
I don't think I can help here. I own a S7-1200 and 1500 but no TIA V16, so someone else need to investigate this :/

MeinzIsAs, 2021-04-30:
Hast Du den Zugriff über PUT/GET Kommunikation in den CPU-Eigenschaften eingeschaltet? 
Schutz & Security --> Verbindungsmechanismen