mathiask88/node-snap7 #77: termux linux aarch64 not compatible！

nmgwr, 2020-12-11:
The Android phone has termux installed and node 14.14.0 installed.

$ uname -a
Linux localhost 4.19.81-perf+ #1 SMP PREEMPT Fri Nov 13 21:32:47 CST 2020 aarch64 Android

Is aarch64 incompatible? Is there any way to be compatible?

error log:

/data/data/com.termux/files/home/web/yitiji-socketserver/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: dlopen failed: "/data/data/com.termux/files/home/web/yitiji-socketserver/node_modules/node-snap7/build/Release/node_snap7.node" has bad ELF magic: cffaedfe
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1122:18)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at bindings (/data/data/com.termux/files/home/web/yitiji-socketserver/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/data/data/com.termux/files/home/web/yitiji-socketserver/node_modules/node-snap7/lib/node-snap7.js:8:45)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Object.require.extensions.<computed> [as .js] (/data/data/com.termux/files/home/web/yitiji-socketserver/node_modules/babel-register/lib/node.js:152:7)

mathiask88, 2020-12-12:
Currently there are no android prebuilds available. Will be available with the NAPI rewrite. If you have a built toolchain on your phone you can try to `npm -i node-snap7 --build-from-source`

nmgwr, 2020-12-13:
What means is built toolchain ？

run： npm -i node-snap7 --build-from-source

[root@localhost yitiji-socketserver]# npm -i node-snap7 --build-from-source

Usage: npm <command>

where <command> is one of:
    access, adduser, audit, bin, bugs, c, cache, ci, cit,
    clean-install, clean-install-test, completion, config,
    create, ddp, dedupe, deprecate, dist-tag, docs, doctor,
    edit, explore, fund, get, help, help-search, hook, i, init,
    install, install-ci-test, install-test, it, link, list, ln,
    login, logout, ls, org, outdated, owner, pack, ping, prefix,
    profile, prune, publish, rb, rebuild, repo, restart, root,
    run, run-script, s, se, search, set, shrinkwrap, star,
    stars, start, stop, t, team, test, token, tst, un,
    uninstall, unpublish, unstar, up, update, v, version, view,
    whoami

npm <command> -h  quick help on <command>
npm -l            display full usage info
npm help <term>   search for help on <term>
npm help npm      involved overview

Specify configs in the ini-formatted file:
    /root/.npmrc
or on the command line via: npm <command> --key value
Config info can be viewed via: npm help config

npm@6.14.4 /usr/lib/node_modules/npm

mathiask88, 2020-12-13:
Install C++11 build tools + python 2.7 to build native node modules and then `npm i node-snap7 --build-from-source` on your target platform (Sorry I had a typo in the last answer)