mathiask88/node-snap7 #92: Won't work with Node.js v20.x

Sinclair81, 2023-10-26:
[SoulL3Zz](https://github.com/SoulL3Zz) posted the following to my [Homebridge plugin](https://github.com/Sinclair81/Homebridge-Logo-Platform):

> Hi, I'm getting an error message since I upgraded to Node.js 20.x:
[10/26/2023, 10:43:53 AM] ====================
[10/26/2023, 10:43:53 AM] ERROR LOADING PLUGIN homebridge-logo-platform:
[10/26/2023, 10:43:53 AM] Error: The module '/var/lib/homebridge/node_modules/homebridge-logo-platform/node_modules/node-snap7/build/Release/node_snap7.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 108. This version of Node.js requires
NODE_MODULE_VERSION 115. Please try re-compiling or re-installing
the module (for instance, using npm rebuild or npm install).
at Object.Module._extensions..node (node:internal/modules/cjs/loader:1327:18)
at Module.load (node:internal/modules/cjs/loader:1091:32)
at Function.Module._load (node:internal/modules/cjs/loader:938:12)
at Module.require (node:internal/modules/cjs/loader:1115:19)
at require (node:internal/modules/helpers:130:18)
at bindings (/var/lib/homebridge/node_modules/homebridge-logo-platform/node_modules/bindings/bindings.js:112:48)
at Object. (/var/lib/homebridge/node_modules/homebridge-logo-platform/node_modules/node-snap7/lib/node-snap7.js:8:45)
at Module._compile (node:internal/modules/cjs/loader:1241:14)
at Object.Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
at Module.load (node:internal/modules/cjs/loader:1091:32)
at Function.Module._load (node:internal/modules/cjs/loader:938:12)
at Module.require (node:internal/modules/cjs/loader:1115:19)
at require (node:internal/modules/helpers:130:18)
at Object. (/var/lib/homebridge/node_modules/homebridge-logo-platform/src/snap7-logo.ts:1:13)
at Module._compile (node:internal/modules/cjs/loader:1241:14)
at Object.Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
[10/26/2023, 10:43:53 AM] ====================

Apollon77, 2023-11-06:
see #93

mathiask88, 2023-11-22:
see #94

closed 2023-11-22 by mathiask88