mathiask88/node-snap7 #97: s7client return a object instead of data

veliadiguzel, 2024-03-27:
Greetings first of all thanks for such a library  i try to read just 1 bit but return object 

`const snap7 = require('node-snap7');
// Bağlantı yapılacak PLC'nin IP adresi ve diğer bağlantı bilgileri
const s7client = new snap7.S7Client();

const ip = '176.16.11.45'; // PLC'nin IP adresi
const rack = 0; // PLC rack numarası
const slot = 3; // PLC slot numarası

s7client.ConnectTo(ip, rack, slot, (err) => {
    if (err) {
        console.error('Connection error: ' + err);
        return;
    }
    else console.log('Connetion established...');

    //    plc.DBRead(215, 256, 1, (err, data) => {
    s7client.ReadArea(s7client.S7AreaDB, 215, 256, 1, s7client.S7WLBit, (err, data) => {

        if (err) {
            console.error('Read Error: ' + err);
            s7client.Disconnect();
            return;
        }
        // Okunan veri DB215.DBX.256.01 adresinde bulunur
        console.log('Read data: ' + data);
        console.log(typeof (data));
        // Bağlantıyı kapat
        s7client.Disconnect();
    });
});`
if i use DBRead and 1 byte get  **%**

PS C:\Users\veli.adiguzel\Desktop\PLC\eaf-status> node .\eaf.js
Connetion established...
Read data: **%**
object      
PS C:\Users\veli.adiguzel\Desktop\PLC\eaf-status> node .\eaf.js
Connetion established...
Read data: 
object     
PS C:\Users\veli.adiguzel\Desktop\PLC\eaf-status>

mathiask88, 2024-03-27:
`ReadArea()` always returns a [buffer ](https://nodejs.org/api/buffer.html)object (For S7WLBit either 0x00 or 0x01). Issue with your code is that you force a type change in your console.log by concatenating the buffer object with a string and 0x00 or 0x01 are no printable utf8 characters.
```
console.log('Read data: ' + data);
```
Try `console.log('Read data: ', data);` instead.

veliadiguzel, 2024-03-28:
I apologize for keeping you busy for something so simple. I'm sorry for switching from Delphi programming language to new internet technologies.

closed 2024-03-28 by veliadiguzel